armink/FlashDB #8: 扇区存在重复擦除的bug

EtcFly, 2020-07-02:
阅读源码的时候, 发现了一个问题：    
      就是的当`sector_iterator(db, &sector, FDB_SECTOR_STORE_UNUSED, &check_failed_count, db, check_sec_hdr_cb, false);` 函数检测到所有的扇区的头部都出问题的时候, 这个时候`check_failed_count`的值应该和该FAL分区中总的扇区数相等, 那么执行      
```
    if (check_failed_count == SECTOR_NUM) {
        FDB_INFO("Warning: All sector header check failed. Set it to default.\n");
        fdb_kv_set_default(db);
    }
```
不就重复的对各扇区多擦除了一次吗？

[image]

其中
```
static bool check_sec_hdr_cb(kv_sec_info_t sector, void *arg1, void *arg2)
{
    if (!sector->check_ok) {
        size_t *failed_count = arg1;
        fdb_kvdb_t db = arg2;

        FDB_INFO("Warning: Sector header check failed. Format this sector (0x%08lX).\n", sector->addr);
        (*failed_count) ++;
        format_sector(db, sector->addr, SECTOR_NOT_COMBINED);
    }

    return false;
}
```
这个函数擦除了一次

armink, 2020-07-02:
这个可能也不算是 bug ，毕竟 set_default 时，不知道已经执行过 format 。set_default 这个 API 用户也可以随时调用

要避免这个重复擦除，可能得再增加额外代码。如果，这个问题对用户产品没有影响，是不是用户也能接受呢。

EtcFly, 2020-07-03:
这确实应该不算bug, 因为这个实际应该只在初始化的时候会出现。  还有一个疑惑, 希望作者可以帮忙解释一下。FlashDB相对于EasyFlash来说, 除了TSDB的时间戳功能, 有哪些性能的提升, 或者功能的迭代？ 对EasyFlash涉世未深,  蛮疑惑

armink, 2020-07-03:
支持 多分区，多实例 。数据量大时，可细化分区，降低检索时间；

这个也算是个最大改变

EtcFly, 2020-07-03:
这个我在阅读源码的时候也发现了, 以前在做easyflash的时候, 如果空间设的较大, 比如1M的时候, 垃圾回收严重到会让外部看门狗复位, 而且数据检索时间延迟较大。 那二者的差异就目前的版本来说这个是主要的功能和性能提升吗？

armink, 2020-07-03:
是的

likyo, 2021-01-26:
我改了一部分代码，让每次只回收一个扇区就退出，下次再回收

closed 2021-10-01 by armink