armink/FlashDB #10: 为什么通过时间查询ts数据库记录，有时能查询到记录，有时候查询不到记录，返回记录数为0

xql923, 2020-07-03:
朱工你好，请问，我通过时间查询ts数据库记录，有时能查询到记录，有时候查询不到记录，返回记录数为0，请问，根据时间查询ts数据库记录时，是否有特殊需要注意的地方？

xql923, 2020-07-03:
你好，我查看源代码：..\FlashDB-master\src\fdb_tsdb.c，在函数fdb_tsl_iter_by_time中，第504行，是否应该加入以下判断语句：
    if(FDB_TSL_WRITE != tsl.status)
	 continue;

   否则，会出现查询不到的情况？

也就是：
do {
                    read_tsl(db, &tsl);
		    if(FDB_TSL_WRITE != tsl.status)
			  continue;
                    if (tsl.time >= from && tsl.time <= to) {
                        /* iterator is interrupted when callback return true */
                        if (cb(&tsl, cb_arg)) {
                            return;
                        }
                    } else {
                        return;
                    }
                } while ((tsl.addr.index = get_next_tsl_addr(&sector, &tsl)) != FAILED_ADDR);

----请问，这样分析对吗？

xql923, 2020-07-03:
只要加上以上两条语句，查询就正常了。

armink, 2020-07-03:
fdb_tsl_iter_by_time 迭代器迭代时候，除了时间没有更多条件

count 统计时，只会统计 FDB_TSL_WRITE  状态的

xql923, 2020-07-03:
但是实际测试时，根据时间检索数据时，发现tsl.status =FDB_TSL_UNUSED时，也进入if (tsl.time >= from && tsl.time <= to)这个语句判断，而这时时间非法，直接return了，所以会查询不到数据。

xql923, 2020-07-03:
请问，这一点，是否我的设置不对，或者用法不对？

xql923, 2020-07-03:
亦或者是其中的一个bug

xql923, 2020-07-03:
如果不加入if(FDB_TSL_WRITE != tsl.status)判断条件，是否可以修改：
if (tsl.time >= from && tsl.time <= to) {
                        /* iterator is interrupted when callback return true */
                        if (cb(&tsl, cb_arg)) {
                            return;
                        }
                    } else {
                        return;  -------------此处修改为 continue;，进行继续迭代循环。
                    }
---实际测试也是可行的。

xql923, 2020-07-03:
朱工，请问，这两种修改方式，从你原本的设计上，是否合理？

armink, 2020-07-03:
提个建议：下次回复问题时，请先多花点时间，想清楚自己要表达的事情，一次性回复好。一句一句的评论不适合在 GitHub 上的交流，还会造成邮件泛滥。

> tsl.status =FDB_TSL_UNUSED时，也进入if (tsl.time >= from && tsl.time <= to)

这个问题是在什么情况下出现的，理论下是不应该有这个状态的 tsl 出现

xql923, 2020-07-03:
好的，朱工！我下次有问题会将所有现象一次性描述清楚。
测试环境：为了测试大数据量存储，数据库存满后，循环存储，我将ts数据库的大小设置为512K Bytes。

         在ts数据初始化后，插入100条数据，查询最近一小时的数据：
        uint32_t curtime = time();
         fdb_tsl_iter_by_time(&MyHs_tsdb, curtime-3600,curtime, query_cb, &cb_arg);
        刚开始都很正常，能正常查询到数据，但是连续插入数据运行一段时间后，查询不到数据。但是可以通过：
        fdb_tsl_iter(&MyHs_tsdb, query_log_cb, &cb_arg);    正常查询到所有数据。

      但是可以通过以上我回复的通过修改fdb_tsl_iter_by_time函数，可以查询到。

      不知道是什么原因？

armink, 2020-07-03:
我这里还不能复现你的问题，我看这两个 API 其实内部差异不大，可否单步调试一下，对比下执行过程的差异呢

xql923, 2020-07-03:
我单步仿真过，通过 read_tsl(db, &tsl);读取一个tsl，它的tsl.status = FDB_TSL_UNUSED，当然时间也就不满足if (tsl.time >= from && tsl.time <= to) 这个条件，因此直接return了，也就是没有查询到数值。

我今天又大数据量测试了，运行一段时间后，就会产生查询不到数据。

请问，我如何提供对你判断问题的有效信息？

xql923, 2020-07-03:
我单步仿真过，通过 read_tsl(db, &tsl);读取一个tsl，它的tsl.status = FDB_TSL_UNUSED，当然时间也就不满足if (tsl.time >= from && tsl.time <= to) 这个条件，因此直接return了，也就是没有查询到数值。

我今天又大数据量测试了，运行一段时间后，就会产生查询不到数据的现象。

请问，我如何提供对你判断问题之做在的有效信息？亦或者您方便的时候，我们远程桌面查看一下？

caiwangz, 2020-07-04:
天龙哥，我也遇到类似问题。
最近在试用你写的FlashDB，感觉很好用，谢谢！ 但现在遇到一个问题，在使用TSDB例程时struct env_status {uint32_t temp; uint32_t humi;}我写入1000个数据，然后进行查询，时间用int数值代替，在不变更存储结构体的情况下，查询范围在[127,127]（或更大的值，单个时间点查询）是没有问题的，但如果把struct env_status {uint32_t temp; uint32_t humi; uint32_t temp2; uint32_t humi2;}结构体数据增加，查询范围在[127,127]（或更大的值，单个时间点查询）查询到的数据个数为0，不知道是不是我哪里使用不正确，希望能够得到你的尽快回复，由衷的感谢！ 
平台：stm32F407，SPI Flash：W25Q128，基于你的FlashDB demo（KVDB已经验证是ok） 

我调试发现如果检索范围是[127,127]，进入fdb_tsl_iter_by_time函数后，tsl.time为127的数据地址不在第一个存储扇区，导致查询出错，如果检索范围是[1,127]，可以检索到所有数据，是因为tsl.time为1的数据在第一个扇区，检束完第一个扇区后，地址通过((tsl.addr.index = get_next_tsl_addr(&sector, &tsl)) != FAILED_ADDR);递增，所以可以继续减速下一个扇区。
不知道是不是这个原因引起的。

xql923, 2020-07-04:
楼上的兄弟，你按照我上述的方法修改一下源代码试试，看能否检索到数据？就是把return修改为continue
如果能检索到数据，我们的问题都是一样的

caiwangz, 2020-07-04:
> 楼上的兄弟，你按照我上述的方法修改一下源代码试试，看能否检索到数据？就是把return修改为continue
> 如果能检索到数据，我们的问题都是一样的

根据你说的方式修改后可以解决。

xql923, 2020-07-04:
那就是这个软件的bug，麻烦朱工解决一下！

caiwangz, 2020-07-04:
使用continue可以解决，但在进行制定范围内的检索还是挺费时间的，是不是可以在这个地方做一个判断，快速进入下一个扇区的搜索，这样可以提升检索的速度。我现在测试单条指令检索，5000多条tsl检索单条，耗时160ms左右。

在2020年07月04日 23:14，xql923 写道：

那就是这个软件的bug，麻烦朱工解决一下！

xql923, 2020-07-05:
那能否麻烦你做出一方案，然后让朱工审核？

armink, 2020-07-05:
@xql923 @caiwangz 你们在移植对接的时候，get_time 的 API 有保证顺序性吗？

就是返回当前时间，中间不能突然比之前最后一次记录 TSL 的时间还早。这点能保证吗？

caiwangz, 2020-07-05:
我们有保障这点。如果写入时间小于上一次会有警告信息提示的。

在2020年07月05日 11:11，朱天龙 (Armink) 写道：

@xql923@caiwangz 你们在移植对接的时候，get_time 的 API 有保证顺序性吗？

就是返回当前时间，中间不能突然比之前最后一次记录 TSL 的时间还早。这点能保证吗？

xql923, 2020-07-05:
> @xql923 @caiwangz 你们在移植对接的时候，get_time 的 API 有保证顺序性吗？
> 
> 就是返回当前时间，中间不能突然比之前最后一次记录 TSL 的时间还早。这点能保证吗？

我没有调整过时间，时间一直增加！

xql923, 2020-07-05:
> @xql923 @caiwangz 你们在移植对接的时候，get_time 的 API 有保证顺序性吗？
> 
> 就是返回当前时间，中间不能突然比之前最后一次记录 TSL 的时间还早。这点能保证吗？

朱工您好，我没有调整时间，取得时间是通过函数time()获取的格林尼治时间，一直在增加。

hcb900330, 2020-07-05:
我也遇到这个问题

xql923, 2020-07-05:
那就是比较普遍的问题了，烦请朱工处理一下！这个问题！

xql923, 2020-07-05:
> 我也遇到这个问题

朱工你好，这个问题应该是个bug引起的！

armink, 2020-07-05:
@xql923 @caiwangz @hcb900330 问题已修正 (https://github.com/armink/FlashDB/commit/f8a3ec152e07b28ed49270a17af053bbe5b3071a) ，请大家试下最新版本

caiwangz, 2020-07-05:
天龙哥，我测试过了，已经修复了！太棒了！现在32000多条tsl进行单条检索，单条检索时间是0.1s左右。

在2020年07月05日 17:04，朱天龙 (Armink) 写道：

@xql923@caiwangz@hcb900330 问题已修正 (f8a3ec1) ，请大家试下最新版本

xql923, 2020-07-05:
楼上的兄弟，请问，您检索到的这些数据接下来是如何处理的？保存成文件吗？

caiwangz, 2020-07-05:
直接用串口打印出来查看。

在2020年07月05日 18:27，xql923 写道：

楼上的兄弟，请问，您检索到的这些数据接下来是如何处理的？保存成文件吗？

armink, 2020-07-05:
没问题了就好，感谢大家的反馈

hcb900330, 2020-07-06:
我的按时间查询还是找不到
finsh />dbtest()
query count is: 0
        0, 0x00000000
finsh />
void dbtest(void)
 { /* QUERY the TSDB by time */
        /* prepare query time (from 1970-01-01 00:00:00 to 2020-05-05 00:00:00) */

        struct tm tm_from = { .tm_year = 2020 - 1900, .tm_mon = 0, .tm_mday = 1, .tm_hour = 0, .tm_min = 0, .tm_sec = 0 };
        struct tm tm_to = { .tm_year = 2020 - 1900, .tm_mon = 6, .tm_mday = 7, .tm_hour = 19, .tm_min =50, .tm_sec = 0 };
        time_t from_time = mktime(&tm_from), to_time = mktime(&tm_to);
        size_t count;
        /* query all TSL in TSDB by time */
        //fdb_time_t from = 0, to = TEST_TS_COUNT -1;

				fdb_tsl_iter_by_time(&tsdb, from_time, to_time, query_cb, NULL);
        /* query all FDB_TSL_WRITE status TSL's count in TSDB by time */
        count = fdb_tsl_query_count(&tsdb, from_time, to_time, FDB_TSL_USER_STATUS1);
        rt_kprintf("query count is: %u\n", count);
 }
 FINSH_FUNCTION_EXPORT(dbtest, dbtest);

但是  全部搜索是可以的
void dbquery(void)
 {  
	 { /* QUERY the TSDB */
        /* query all TSL in TSDB by iterator */
        fdb_tsl_iter(&tsdb, query_cb, &tsdb);
   }
 }
 FINSH_FUNCTION_EXPORT(dbquery, dbquery);
ueried a TSL: time: 1594026260, event: ±¸µç¹ÊÕÏ, position:
queried a TSL: time: 1594026260, event: µØÖ·1 Í¨µÀ1ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·1 Í¨µÀ2ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·1 Í¨µÀ3ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·1 Í¨µÀ4ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·1 Í¨µÀ5ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·1 Í¨µÀ6ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·2 Í¨µÀ1ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·2 Í¨µÀ2ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·2 Í¨µÀ3ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·2 Í¨µÀ4ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·2 Í¨µÀ5ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·2 Í¨µÀ6ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·3 Í¨µÀ1ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·3 Í¨µÀ2ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·3 Í¨µÀ3ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·3 Í¨µÀ4ÖÐ¶Ï, position:
queried a TSL: time: 1594026260, event: µØÖ·3 Í¨µÀ5ÖÐ¶Ï, position:
。。。。。。。。。。。

我感觉时间戳没有问题是这样写的
static fdb_time_t get_time(void)
{
    time_t now;

    now = time(RT_NULL);
    return now;
}

hcb900330, 2020-07-06:
struct Status {
    char  event[64];
	char  position[64];
};

我添加的数据格式是这样的，请大家帮忙看看

armink, 2020-07-06:
其他人还有这个问题吗？

@hcb900330 找时间我再帮你看一下

armink, 2020-07-06:
@hcb900330 针对 fdb_tsl_iter_by_time 的 from 查询条件又做了一些完善(https://github.com/armink/FlashDB/commit/8fb75abea0730581ed14c704b94a8a3e3e8d1ff4) ，你再更新试一下吧

caiwangz, 2020-07-06:
我也安排测试一下哈

在2020年07月06日 22:55，朱天龙 (Armink) 写道：

@hcb900330 针对 fdb_tsl_iter_by_time 的 from 查询条件又做了一些完善(8fb75ab) ，你再更新试一下吧

hcb900330, 2020-07-07:
我测试了下  按时间在我这里还有有问题
[E/FAL] (fal_partition_read:389) Partition read error! Don't found flash device(ÿ½µð{ÿ%ð©þðÿ½µðsÿ) of the partition(ç).

全部查询还是可以的  

另外我的fal好像有问题  之前我改了下分区空间，使用的还是0.2，改成最新的再看看
000000 | len: 0x01000000 | blk_size: 0x00001000 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name       | flash_dev  |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | easyflash  | nor_flash0 | 0x00000000 | 0x00100000 |
[I/FAL] | fdb_kvdb1  | nor_flash0 | 0x00400000 | 0x00100000 |
[I/FAL] | filesystem | nor_flash0 | 0x00600000 | 0x00300000 |
[I/FAL] | sqlite     | nor_flash0 | 0x00a00000 | 0x00200000 |
[I/FAL] | fdb_tsdb1  | nor_flash0 | 0x00d00000 | 0x00300000 |
[I/FAL] =============================================================

[E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound.
[E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound.
[E/FAL] (fal_partition_erase:462) Partition erase error! Partition address out of bound.

caiwangz, 2020-07-07:
我原来出现类似问题，后来发现是我在query_cb函数时使用不当，读取参数的输入不对。我的现象是查询地址本应该在0x800000后面，但因为函数使用不当，导致查询地址跑到了0x30，
提示(init ok)..Don’t found flash device of the partition
在2020年07月07日 08:34，hcb900330 写道：

我测试了下 按时间在我这里还有有问题
[E/FAL] (fal_partition_read:389) Partition read error! Don't found flash device(ÿ½µð{ÿ%ð©þðÿ½µðsÿ) of the partition(ç).

全部查询还是可以的

另外我的fal好像有问题 之前我改了下分区空间，使用的还是0.2，改成最新的再看看
000000 | len: 0x01000000 | blk_size: 0x00001000 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name | flash_dev | offset | length |
[I/FAL] -------------------------------------------------------------
[I/FAL] | easyflash | nor_flash0 | 0x00000000 | 0x00100000 |
[I/FAL] | fdb_kvdb1 | nor_flash0 | 0x00400000 | 0x00100000 |
[I/FAL] | filesystem | nor_flash0 | 0x00600000 | 0x00300000 |
[I/FAL] | sqlite | nor_flash0 | 0x00a00000 | 0x00200000 |
[I/FAL] | fdb_tsdb1 | nor_flash0 | 0x00d00000 | 0x00300000 |
[I/FAL] =============================================================

[E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound.
[E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound.
[E/FAL] (fal_partition_erase:462) Partition erase error! Partition address out of bound.

hcb900330, 2020-07-07:
我使用的外部的spiflash，查询函数也是照抄的
static bool query_cb1(fdb_tsl_t tsl, void *arg)
{
    struct fdb_blob blob;
    struct Status status;
    fdb_tsdb_t db = arg;
    //fdb_blob_make(&blob, data, tsl->log_len);
    fdb_blob_read((fdb_db_t) db, fdb_tsl_to_blob(tsl, fdb_blob_make(&blob, &status, sizeof(status))));
    rt_kprintf("queried a TSL: time: %d, event: %s, position: %s\n", tsl->time, status.event, status.position);

    return false;
}

armink, 2020-07-07:
@hcb900330 你现在的问题是 `Partition read error! Partition address out of bound` 吗？如果是的话，请独立一个 issue

hcb900330, 2020-07-07:
> 我测试了下 按时间在我这里还有有问题
> [E/FAL] (fal_partition_read:389) Partition read error! Don't found flash device(ÿ½µð{ÿ%ð©þðÿ½µðsÿ) of the partition(ç).
>
> 全部查询还是可以的
>
> 另外我的fal好像有问题 之前我改了下分区空间，使用的还是0.2，改成最新的再看看 [...]

使用了最新的fal，还是一样的，fdb_tsl_query_count(&tsdb, from_time, to_time, FDB_TSL_WRITE);返回数目是正确的，就是显示内容不对fdb_tsl_iter_by_time(&tsdb, from_time, to_time, query_cb1, NULL);

void dbtest(void)
 { /* QUERY the TSDB by time */
        /* prepare query time (from 1970-01-01 00:00:00 to 2020-05-05 00:00:00) */
//        struct tm tm_from = { .tm_year = 1970 - 1900, .tm_mon = 0, .tm_mday = 1, .tm_hour = 0, .tm_min = 0, .tm_sec = 0 };
//        struct tm tm_to = { .tm_year = 2020 - 1900, .tm_mon = 4, .tm_mday = 5, .tm_hour = 0, .tm_min = 0, .tm_sec = 0 };
	 
        struct tm tm_from = { .tm_year = 2020 - 1900, .tm_mon = 0, .tm_mday = 1, .tm_hour = 0, .tm_min = 0, .tm_sec = 0 };
        struct tm tm_to = { .tm_year = 2020 - 1900, .tm_mon = 6, .tm_mday = 8, .tm_hour = 19, .tm_min =50, .tm_sec = 0 };
        time_t from_time = mktime(&tm_from), to_time = mktime(&tm_to);
        size_t count;
        /* query all TSL in TSDB by time */
        //fdb_time_t from = 0, to = TEST_TS_COUNT -1;

				fdb_tsl_iter_by_time(&tsdb, from_time, to_time, query_cb1, NULL);
        /* query all FDB_TSL_WRITE status TSL's count in TSDB by time */
        count = fdb_tsl_query_count(&tsdb, from_time, to_time, FDB_TSL_WRITE);
        rt_kprintf("query count is: %u\n", count);
 }
 FINSH_FUNCTION_EXPORT(dbtest, dbtest);

caiwangz, 2020-07-07:
建议先完全按照天龙哥提供的例程测试后，再修改，防止多问题耦合。

在2020年07月07日 09:18，hcb900330 写道：

我原来出现类似问题，后来发现是我在query_cb函数时使用不当，读取参数的输入不对。我的现象是查询地址本应该在0x800000后面，但因为函数使用不当，导致查询地址跑到了0x30， 提示(init ok)..Don’t found flash device of the partition 在2020年07月07日 08:34，hcb900330 写道： 我测试了下 按时间在我这里还有有问题 [E/FAL] (fal_partition_read:389) Partition read error! Don't found flash device(ÿ½µð{ÿ%ð©þðÿ½µðsÿ) of the partition(ç). 全部查询还是可以的 另外我的fal好像有问题 之前我改了下分区空间，使用的还是0.2，改成最新的再看看 000000 | len: 0x01000000 | blk_size: 0x00001000 |initialized finish. [I/FAL] ==================== FAL partition table ==================== [I/FAL] | name | flash_dev | offset | length | [I/FAL] ------------------------------------------------------------- [I/FAL] | easyflash | nor_flash0 | 0x00000000 | 0x00100000 | [I/FAL] | fdb_kvdb1 | nor_flash0 | 0x00400000 | 0x00100000 | [I/FAL] | filesystem | nor_flash0 | 0x00600000 | 0x00300000 | [I/FAL] | sqlite | nor_flash0 | 0x00a00000 | 0x00200000 | [I/FAL] | fdb_tsdb1 | nor_flash0 | 0x00d00000 | 0x00300000 | [I/FAL] ============================================================= [E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound. [E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound. [E/FAL] (fal_partition_erase:462) Partition erase error! Partition address out of bound.

我使用的外部的spiflash，查询函数也是照抄的
static bool query_cb1(fdb_tsl_t tsl, void *arg)
{
struct fdb_blob blob;
struct Status status;
fdb_tsdb_t db = arg;
//fdb_blob_make(&blob, data, tsl->log_len);
fdb_blob_read((fdb_db_t) db, fdb_tsl_to_blob(tsl, fdb_blob_make(&blob, &status, sizeof(status))));
rt_kprintf("queried a TSL: time: %d, event: %s, position: %s\n", tsl->time, status.event, status.position);

return false;

}

hcb900330, 2020-07-07:
> @hcb900330 你现在的问题是 `Partition read error! Partition address out of bound` 吗？如果是的话，请独立一个 issue

fal没有问题 ，我改过来了，我还换了一个全新的一个spiflash  想着之前是不是我修改分区控件错乱了，还是不行。
全局搜索就不会有问题      

按时间搜索就会出现这个问题，应该跟fal没关系

E/FAL] (fal_partition_read:389) Partition read error! Don't found flash device(ÿ½µð{ÿ%ð©þðÿ½µðsÿ) of the partition(ç).
Ìh ¬h , position: $Â 1594119004, event:

xql923, 2020-07-07:
朱工你好，这一个功能确实有问题，你修改的最后版本，我又查询不到数据了。

xql923, 2020-07-07:
@caiwangz，请你也测试一下！

hcb900330, 2020-07-07:
全部删除记录以后也会出问题
[E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound.
[E/FAL] (fal_partition_read:382) Partition read error! Partition address out of bound.

caiwangz, 2020-07-07:
刚才用天龙哥6号更新的程序，测试了两个，一个是查询之前写在SPI FLASH的程序，是正常的。后面我又格式化后，写一次查询一次都是正常的。我测试的时间是每次写一个数据累加1，同时写入到kvdb，上电再读取这个值，保证连续性。 
综合来说，目前测试正常的。

在2020年07月07日 12:41，xql923 写道：

@caiwangz，请你也测试一下！

xql923, 2020-07-07:
谢谢！

hcb900330, 2020-07-08:
fdb  ts本身就不要考虑时间的问题吧，我这样添加有问题么？
static void rev_history_entry(void* parameter)
{
    char buf[160];
	  struct fdb_blob blob;
    while (1)
    {
        rt_memset(&buf[0], 0, sizeof(SMSG));

        if(rt_mq_recv(&histroy_mq, &buf[0], sizeof(SMSG), RT_WAITING_FOREVER) == RT_EOK)
        {
					  struct Status  status;
					  SMSG *smsg=(SMSG *)buf;
					  rt_memset(status.event,0,64);
					  rt_memset(status.position,0,64);
            rt_memcpy(status.event,smsg->event,64);
					  rt_memcpy(status.position,smsg->position,64);
						fdb_tsl_append(&tsdb, fdb_blob_make(&blob, &status, sizeof(status)));
						//rt_kprintf("append the new status.event (%s) and status.position (%s)\n", status.event, status.position);
	               
        }
				
		

    }
}

hcb900330, 2020-07-08:
不同的线程之间操作fdb  我并没有加互斥锁，会不会这个问题导致出问题
static void lock(fdb_db_t db)
{
    __disable_irq();
}

static void unlock(fdb_db_t db)
{
    __enable_irq();
}

static fdb_time_t get_time(void)
{
    time_t now;

    now = time(RT_NULL);
    return now;
}

closed 2021-02-28 by armink