armink/FlashDB #11: KV iterator export to API

eggcar, 2020-07-03:
能否添加一个公开的KV的iterator api？
目前需要手动暴露内部的kv_iterator()接口来遍历整个db，比较麻烦而且需要暴露库的内部实现

可以考虑添加一个迭代器结构体，用迭代器结构体来存储状态，然后api每次返回一个迭代结果，这样用起来更方便一些

armink, 2020-07-03:
是有可以计划公开  KV 迭代器的 API ，或者你来实现一下？完事直接 PR 上来

eggcar, 2020-07-03:
好滴，我先做一版提个PR

armink, 2020-07-03:
fixed in https://github.com/armink/FlashDB/pull/12

xql923, 2020-07-06:
请检查源程序..\FlashDB-master\src\fdb_kvdb.c中第1673行：kv->addr.start == 0;
该语句是否写错了？应该为：
kv->addr.start = 0;

eggcar, 2020-07-06:
> 
> 
> 请检查源程序..\FlashDB-master\src\fdb_kvdb.c中第1673行：kv->addr.start == 0;
> 该语句是否写错了？应该为：
> kv->addr.start = 0;

已修复。晚些时候合并一个PR过来

xql923, 2020-07-06:
还有，你修改后的，能否这里出一个完整的例程，说明如何使用，可以吗？现在有点不太明白。

xql923, 2020-07-06:
还有，你修改后的，能否整理出一份完整的使用例程，说明如何使用，可以吗？现在有点不太明白你的用意。

eggcar, 2020-07-06:
> 
> 
> 还有，你修改后的，能否整理出一份完整的使用例程，说明如何使用，可以吗？现在有点不太明白你的用意。

要遍历kvdb，首先定义一个迭代器对象，然后调用初始化函数对迭代器对象进行初始化：

```
struct fdb_kv_iterator iterator;
fdb_kv_iterator_init(&iterator);
```

然后，用一个while循环来调用fdb_kv_iterate()函数来执行遍历：

```
fdb_kv_t		itr_kv;
while (fdb_kv_iterate(env_db, &iterator)) {
		itr_kv = &(iterator.curr_kv);
		data_size = (size_t)itr_kv->value_len;
		data_buf = sh_malloc(data_size);
		if (data_buf == NULL) {
			sh_printf(ofd, "Malloc failed.\r\n");
			return -ENOMEM;
		}
		fdb_blob_read(env_db, fdb_kv_to_blob(itr_kv, fdb_blob_make(&blob, data_buf, data_size)));
		/* blahblahblah do what ever you like with blob... */
		sh_free(data_buf);
}
```

迭代未结束，函数返回true，while一直循环，直到遍历结束，函数返回false，退出循环

迭代器对象里有三个统计值，统计**当前**迭代得到有效的键值对个数、有效的value占用空间、有效的kv对象占用空间。
注意，**每次遍历之前必须对迭代器对象执行初始化**。**不要在你的应用代码里手动修改迭代器对象的内容**，尤其是sector_addr成员。

xql923, 2020-07-06:
谢谢！

closed 2021-02-28 by armink

lilealyh, 2022-08-11:
> > 还有，你修改后的，能否整理出一份完整的使用例程，说明如何使用，可以吗？现在有点不太明白你的用意。
> 
> 要遍历kvdb，首先定义一个迭代器对象，然后调用初始化函数对迭代器对象进行初始化：
> 
> ```
> struct fdb_kv_iterator iterator;
> fdb_kv_iterator_init(&iterator);
> ```
> 
> 然后，用一个while循环来调用fdb_kv_iterate()函数来执行遍历：
> 
> ```
> fdb_kv_t		itr_kv;
> while (fdb_kv_iterate(env_db, &iterator)) {
> 		itr_kv = &(iterator.curr_kv);
> 		data_size = (size_t)itr_kv->value_len;
> 		data_buf = sh_malloc(data_size);
> 		if (data_buf == NULL) {
> 			sh_printf(ofd, "Malloc failed.\r\n");
> 			return -ENOMEM;
> 		}
> 		fdb_blob_read(&(env_db->parent), fdb_kv_to_blob(itr_kv, fdb_blob_make(&blob, data_buf, data_size)));
> 		/* blahblahblah do what ever you like with blob... */
> 		sh_free(data_buf);
> }
> ```
请问下遍历后itr_kv->name 打印一直是空的 没有什么内容是什么原因？

eggcar, 2022-11-01:
> 

@lilealyh 请问是使用的仓库最新的代码吗？itr_kv->name_len是否为0呢？
也看下iterator.iterated_cnt，这个代表当前遍历到的有效的k-v对的数量，如果一直是0，是不是kvdb本身是空的？

xql923, 2022-11-01:
> @lilealyh 请问是使用的仓库最新的代码吗？itr_kv-&gt;name_len是否为0呢？

建议你阅读一下FlashDB的源代码，非常精妙！