armink/FlashDB #9: ts数据库存满后如何处理？

xql923, 2020-07-03:
你好，请问，对于一个ts数据库，如果数据存满空间后，会如何处理？删除最早的数据而插入最新的数据吗

armink, 2020-07-03:
取决于数据库的设置，设置为滚动保存时，会覆盖最早的，否则就提示插入失败了，需要用户手动清空数据库

详见 control API  https://github.com/armink/FlashDB/blob/master/src/fdb_tsdb.c#L671

xql923, 2020-07-03:
谢谢，请问这个软件有详细的使用说明书吗？

xql923, 2020-07-03:
请问，fdb_tsl_query_count(fdb_tsdb_t db, fdb_time_t from, fdb_time_t to, fdb_tsl_status_t status)，这个函数中，fdb_tsl_status_t status主要做什么作用？

xql923, 2020-07-03:
我看说明上修改状态：
NOTE: The actions to modify the state must be in order.
         *       FDB_TSL_WRITE -> FDB_TSL_USER_STATUS1 -> FDB_TSL_DELETED -> FDB_TSL_USER_STATUS2
这些状态位，必须顺序修改吗？

armink, 2020-07-03:
更多的文档在准备中，可以先看一下 sample 里的示例，快速学习一下

xql923, 2020-07-03:
谢谢！正在阅读源代码

yolee599, 2020-12-23:
TSDB 启用滚动写入，当可用扇区写满的时候，是不是最早的一整个扇区的数据都会被清除，而不是一条一条清除？

closed 2021-10-01 by armink