armink/FlashDB #16: 如果时钟向前调整，比如向前1个小时，那FlashDB该如何处理呢？

xql923, 2020-07-06:
朱工您好，按照你的提示：就是入库记录时间不能比上一个记录的时间小，那如果系统时钟错误，就必须向前调整，比如当前是10：00，调整到9：00，那么就会出现当前的记录时间比上一个记录的时间小，请问，程序应该如何处理呢？

armink, 2020-07-06:
程序一般做法为删除之前的 9:00-10:00 的数据

eggcar, 2020-07-06:
建议用单调时钟作为时间戳，不要直接用系统时间

xql923, 2020-07-06:
> 程序一般做法为删除之前的 9:00-10:00 的数据

如何删除之前的数据？有提供api吗？

xql923, 2020-07-06:
> 程序一般做法为删除之前的 9:00-10:00 的数据

如果用所谓的单调时间而不是系统时间，那么通过数据的时间查询就没有意义了

xql923, 2020-07-06:
> 程序一般做法为删除之前的 9:00-10:00 的数据

按照你之前的回答，不能直接删除数据，所谓删除只能是修改tsl的状态，是吗？

armink, 2020-07-07:
> > 程序一般做法为删除之前的 9:00-10:00 的数据
> 
> 按照你之前的回答，不能直接删除数据，所谓删除只能是修改tsl的状态，是吗？

是的，改为删除状态，应用用的时候就可以有所区分

xql923, 2020-07-07:
谢谢朱工！

xql923, 2020-07-07:
朱工你好，请问，是否是这样做：
static bool set_status_cb(fdb_tsl_t tsl, void *arg)
{
    fdb_tsdb_t db = arg;

    FDB_INFO("set the TSL (time %ld) status from %d to %d\n", tsl->time, tsl->status, FDB_TSL_USER_STATUS1);
    fdb_tsl_set_status(db, tsl, FDB_TSL_USER_STATUS1);

    return false;
}

static void delete_preTime_hisData(uint32_t tsl_time,uint8_t datatype)
{
	uint32_t Last_tsl_time = 0;

	/* read last saved time for simulated timestamp */
	 fdb_tsdb_control(&RainHs_tsdb, FDB_TSDB_CTRL_GET_LAST_TIME, &Last_tsl_time);
	if (tsl_time < Last_tsl_time)
	{
		fdb_tsl_iter_by_time(&RainHs_tsdb, tsl_time-1, Last_tsl_time, set_status_cb, &RainHs_tsdb);
	}	
}

然后在每次运行插入tsl前，首先判断时间，如果小于上一条记录时间，就删除：
delete_preTime_hisData(RainHs_tsdb.get_time(),Rain);
result = fdb_tsl_append(&RainHs_tsdb, fdb_blob_make(&blob, p_element, sizeof(hrv_history_key)));

---请问，朱工，以上的流程对吗？

xql923, 2020-07-07:
另外一个问题，如果不删除之前的数据会导致什么后果？因为如果是日志信息，所有的我都不想删除。

armink, 2020-07-08:
- 1、删除不是用 `FDB_TSL_USER_STATUS1` 而是用 `FDB_TSL_DELETE`
- 2、在每次运行插入tsl前都判断会不会有些繁琐，第一次启动时候判断下应该就行了
- 3、不删除会导致数据库时间轴出现紊乱，查询的结果也会乱套
- 4、如果只是日志，可以不用设置 RTC 时间戳，参考我前两天新改的 demo，每次启动时，读取上次保存的 last_time ，这样是可以保证记录的有序性

xql923, 2020-07-08:
朱工你好，谢谢你的回复，你在例程中这样写道：
* NOTE: The actions to modify the state must be in order.
         *       FDB_TSL_WRITE -> FDB_TSL_USER_STATUS1 -> FDB_TSL_DELETED -> FDB_TSL_USER_STATUS2
是否必须按照这个顺序来设置TSL的状态，比如需要设置tsl为 FDB_TSL_DELETED ，首先要设置为FDB_TSL_USER_STATUS1，再次设置FDB_TSL_DELETED？

armink, 2020-07-08:
是得按照顺序，但可以跳过中间状态

xql923, 2020-07-08:
嗯，好的，谢谢朱工！
FlashDB很好用，在其进行初始化时，需要对每一个secotr写一个控制信息，如果分区空间比较大，比如3M Bytes，那么初始化时间就比较长，有没有办法改进一下，以减少初始化时间？

armink, 2020-07-08:
第一次格式化的时间好像没法省吧，这个也就只有第一次才存在问题

xql923, 2020-07-08:
> 第一次格式化的时间好像没法省吧，这个也就只有第一次才存在问题

谢谢朱工，谢谢！

cainiaoPro, 2024-08-12:
@armink 朱工你好，请问下我这边也遇到该问题，但是发现通过fdb_tsl_set_status(db, tsl, FDB_TSL_DELETED);将调整过时间的记录改为FDB_TSL_DELETED状态后发现db->last_time的值没有更新，还是被删除的那条的值