armink/FlashDB #21: 关于分区起始地址错误问题

ZakiLiu, 2020-07-14:
## 硬件软件环境
 1. Hardware
    MCU： STM32F103VC
    norflash：W25Q128

2. Software
    IDE：MDK 5.30.0.0
    C-Complier： ArmClang V6.14
    RTOS： FreeRTOS V10.3.1
## 问题
1. 关于kvdb测试发现，kvdb分区最好小于1M，不然初始化会使CPU 100%运行500ms以上（如下图），主要是由于迭代器函数sector_iterator()导致的，这部分可能需要算法优化吧。

[image] 

2. 关于kvdb和tsdb分区问题，测试发现，kvdb（一般比较小）最好放在tsdb的前面，这样kvdb的初始化耗时少，通过打印迭代器函数的addr发现了一些端倪，比如测试分区如下：
[image]
kvdb的起始地址是0x00200000，但打印迭代器的初始化地址是0x00000000；也就是说迭代器是从tsdb的起始地址查询到kvdb的结束地址0x00202000；我通过修改以下代码（注释部分为原始代码），可以较少迭代次数，但flash重新格式化后会有“**Partition read error! Partition address out of bound.**”错误，因为对flashDB的逻辑不清楚，所以又将代码还原。
```c
static void kv_iterator(fdb_kvdb_t db, fdb_kv_t kv, void *arg1, void *arg2,
        bool (*callback)(fdb_kv_t kv, void *arg1, void *arg2))
{
    struct kvdb_sec_info sector;
    uint32_t sec_addr;

    // sec_addr = 0;    
    sec_addr = (uint32_t)((fdb_db_t)db)->part->offset;
```

```c
static void sector_iterator(fdb_kvdb_t db, kv_sec_info_t sector, fdb_sector_store_status_t status, void *arg1, void *arg2,
        bool (*callback)(kv_sec_info_t sector, void *arg1, void *arg2), bool traversal_kv)
{
    uint32_t sec_addr;

    /* search all sectors */
    // sec_addr = 0;   
    sec_addr = (uint32_t)((fdb_db_t)db)->part->offset;
```

ZakiLiu, 2020-07-14:
是我把fal的分区表理解错了

armink, 2020-07-14:
OK，那我先关闭了哈

closed 2020-07-14 by armink