armink/FlashDB #23: 能否添加SPI FLASH 多分区同时操作的样例

1XiaSheng, 2020-07-20:
在使用SPI FLASH情况下啊，已经存在多个分区，怎样区别操作不同分区？
在fal_flash_sfud_port.c 文件中，
···
struct fal_flash_dev nor_flash0 =
{
    .name       = FAL_USING_NOR_FLASH_DEV_NAME,
    .addr       = 0,
    .len        = 4*1024*1024,
    .blk_size   = 4096,
    .ops        = {init, read, write, erase},
    .write_gran = 1
};
···
中的 addr 和 len 修改后可否限制操作到其他位置，例如 addr = 10240； len = 10240， 是否说明 地址0-10239 和 20480之后的空间是绝对不会用到的。
分区表中的地址是在此定义空间中相对划分的吗？

1XiaSheng, 2020-07-20:
在 fdb_kv_print时 [fdb_kvdb.c:1347] blob @0x0003A1AA 160bytes ，kv->addr.value 存储的时绝对地址还是相对地址？

armink, 2020-07-20:
这个 demo 里就是多分区的 https://github.com/armink/FlashDB/blob/master/demos/stm32f405rg_spi_flash/applications/fal_cfg.h

分区内部只用关心相对地址，不关心完整的物理绝对地址

1XiaSheng, 2020-07-24:
还是自己摸出来了

closed 2020-07-24 by 1XiaSheng