armink/FlashDB #25: 对于键值KV型数据库，每个键值支持最大的字符数目是多少？

xql923, 2020-08-01:
朱工你好，请问，对于键值KV型数据库，每个键值支持最大的字符数目是多少？

armink, 2020-08-01:
在这里有配置，可以直接在 fdb_cfg.h 中定义一下

https://github.com/armink/FlashDB/blob/master/src/fdb_kvdb.c#L51

xql923, 2020-08-01:
谢谢朱工，请问设置这个数值很大或者很小对数据库的性能有影响吗？

xql923, 2020-08-01:
比如我的Nor Flash是W25Q256，一个页Page大小是256Bytes，一个扇区Sector大小是4096Bytes，设置FDB_STR_KV_VALUE_MAX_SIZE为多大最能发挥数据库的效能和Flash的最大性能？

armink, 2020-08-02:
其实这个对性能影响不大，就是会多占用一些内存

hcb900330, 2021-01-29:
我查看了下， 您这个使用栈里面的内存， 我现在想把内存扩大点，500k     我使用了堆里面的内存   rt_malloc          修改了下面这个函数，也不报错了，还是存不进去  ， set-kv那里的代码不知道怎么修改  
char *fdb_kv_get(fdb_kvdb_t db, const char *key)
{
    //static char value[FDB_STR_KV_VALUE_MAX_SIZE + 1];
    char* value=rt_malloc(FDB_STR_KV_VALUE_MAX_SIZE + 1)
    size_t get_size;
    struct fdb_blob blob;

    if ((get_size = fdb_kv_get_blob(db, key, fdb_blob_make(&blob, value, FDB_STR_KV_VALUE_MAX_SIZE))) > 0) {
        /* the return value must be string */
        //if (fdb_is_str((uint8_t *)value, get_size))

       {
            value[get_size] = '\0';
            return value;
        } 
    }

hcb900330, 2021-01-29:
> 其实这个对性能影响不大，就是会多占用一些内存
static  char kv_kv[1024*1024] __attribute__( (at(0xC0F00000)) ); 
char *fdb_kv_get(fdb_kvdb_t db, const char *key)
{
    //static char value[FDB_STR_KV_VALUE_MAX_SIZE + 1];
	  char *value=kv_kv;
    size_t get_size;
    struct fdb_blob blob;

    if ((get_size = fdb_kv_get_blob(db, key, fdb_blob_make(&blob, value, FDB_STR_KV_VALUE_MAX_SIZE))) > 0) {
        /* the return value must be string */
        if (fdb_is_str((uint8_t *)value, get_size)) {
            value[get_size] = '\0';
            return value;
        } else if (blob.saved.len > FDB_STR_KV_VALUE_MAX_SIZE) {
            FDB_INFO("Warning: The default string KV value buffer length (%d) is too less (%zu).\n", FDB_STR_KV_VALUE_MAX_SIZE,
                    blob.saved.len);
        } else {
            FDB_INFO("Warning: The KV value isn't string. Could not be returned\n");
            return NULL;
        }
    }

我这样修改下还是不行

hcb900330, 2021-01-29:
不修改任何源码，#define FDB_STR_KV_VALUE_MAX_SIZE               1024*120    取value也取不出来

打印
[FlashDB][kv][env] Warning: The KV value isn't string. Could not be returned
[FlashDB][kv][env] Warning: The KV value isn't string. Could not be returned
[FlashDB][kv][env] Warning: The KV value isn't string. Could not be returned

closed 2021-10-01 by armink