armink/FlashDB #28: 是否能添加对Linux等基于文件的db文件支持？

jjldc, 2020-08-26:


armink, 2020-08-26:
可以考虑的。能大概说下你们的应用场景和具体需求吗？

jjldc, 2020-08-26:
我们希望在除了RTT之外，在一些跑Linux或其他标准OS的产品上使用类似的时序和KV数据库。
flash损耗均衡等问题，都由操作系统或底层SSD控制器处理，对DB来说其实很简单，专注数据层面的问题即可，这样可以将FlashDB的应用扩展到更多高端的产品上

armink, 2020-08-26:
其实在现在版本 FlashDB 可以通过文件模拟 Flash 方式运行起来

在 FlashDB 的 GitHub CI 里，就已经使用文件系统作为存储介质，自动化运行 FlashDB 里的一些测试用例，你可以参考一下

当然，也建议先从应用多用用 FlashDB ，看下能够满足产品上业务要求

donny681, 2020-10-26:
我测试过，linux下支持的
`
/*
 * Copyright (c) 2020, Armink, <armink.ztl@gmail.com>
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <string.h>
#include <fal.h>

#define FLASH_FILE_PATH       "./flash_sim.bin"
#define FLASH_FILE_SIZE       (1*1024*1024)
#define FLASH_FILE_PAGE_SIZE  (4*1024)
static FILE *flash_fp = NULL;

static int init(void)
{

    flash_fp = fopen(FLASH_FILE_PATH, "ab+");
    if (flash_fp) {
        
        fclose(flash_fp);
    }else{
        printf("fopen err\r\n");
    }

    return 0;
}

static int read(long offset, uint8_t *buf, size_t size)
{
    //Xassert(size % 4 == 0);

    /* You can add your code under here. */
    flash_fp = fopen(FLASH_FILE_PATH, "rb+");
    if (flash_fp) {
        
        fseek(flash_fp, offset, SEEK_SET);
        fread(buf, size, 1, flash_fp);
        fclose(flash_fp);
        return size;
    } else {
        printf("fread err\r\n");
        return -1;
    }
}

static int write(long offset, const uint8_t *buf, size_t size)
{
    flash_fp = fopen(FLASH_FILE_PATH, "rb+");

    if (flash_fp){
        
        fseek(flash_fp, offset, SEEK_SET);
        fwrite(buf, size, 1, flash_fp);
        fclose(flash_fp);
        return size;
    } else {
         printf("fwrite err\r\n");
        return -1;
    }
}

static int erase(long offset, size_t size)
{
#define BUF_SIZE 32

    uint8_t buf[BUF_SIZE];
    size_t i;

    flash_fp = fopen(FLASH_FILE_PATH, "rb+");

    if (flash_fp) {
         printf("erase err\r\n");
        fseek(flash_fp, offset, SEEK_SET);
        for (i = 0; i * BUF_SIZE < size; i++)
        {
            memset(buf, 0xFF, BUF_SIZE);
            fwrite(buf, BUF_SIZE, 1, flash_fp);
        }

        memset(buf, 0xFF, BUF_SIZE);
        fwrite(buf, size - i * BUF_SIZE, 1, flash_fp);
        fclose(flash_fp);

        return size;
    } else {
        return -1;
    }
}

/*
  "esp8266_onchip" : Flash 设备的名字。
  0x08000000: 对 Flash 操作的起始地址。
  1024*1024：Flash 的总大小（1MB）。
  128*1024：Flash 块/扇区大小（因为 STM32F2 各块大小不均匀，所以擦除粒度为最大块的大小：128K）。
  {init, read, write, erase} ：Flash 的操作函数。 如果没有 init 初始化过程，第一个操作函数位置可以置空。
  8 : 设置写粒度，单位 bit， 0 表示未生效（默认值为 0 ），该成员是 fal 版本大于 0.4.0 的新增成员。各个 flash 写入粒度不尽相同，可通过该成员进行设置，以下列举几种常见 Flash 写粒度：
  nor flash:  1 bit
  stm32f2/f4: 8 bit
  stm32f1:    32 bit
  stm32l4:    64 bit
 */

//1.定义 flash 设备

const struct fal_flash_dev nor_flash0 =
{
    .name       = "norflash0",
    .addr       = 0,
    .len        = FLASH_FILE_SIZE,
    .blk_size   = FLASH_FILE_PAGE_SIZE,
    .ops        = {init, read, write, erase},
};
`

jjldc, 2020-12-28:
@armink 朱工我看到autoTest项目中正在做POSIX_MODE开发，请问这块是否可以使用了？

armink, 2020-12-28:
是的，功能上都可以用了，现在在做一些深入的测试，晚些时候会推送上来

armink, 2021-01-03:
文件模式已支持，可以用 linux 下的 demo 先尝尝鲜哈 https://github.com/armink/FlashDB/tree/master/demos/linux

eleree, 2021-04-06:
请问下现在master分支上支持linux了吗？目前手动修改编译完后直接
Floating point exception (core dumped)

armink, 2021-04-06:
> 请问下现在master分支上支持linux了吗？目前手动修改编译完后直接
> Floating point exception (core dumped)

支持的，可以直接编译、运行的