armink/FlashDB #29: 关于KVDB，没有在default_kv中加入kv值，而是在运行中存储kv值，如果运行中的kv值改变，会导致数值读取错误吗？

ZakiLiu, 2020-08-26:
具体的应用如下：
初始化KVDB
```c
static struct fdb_default_kv_node default_kv_table[] = {
    {"boot_count", &boot_count, sizeof(boot_count)}, 
    {"boot_time", &boot_time, sizeof(boot_time)},  
}
// 初始化函数
void flash_kv_init(void)
{
   struct fdb_default_kv default_kv;

   default_kv.kvs = default_kv_table;
   default_kv.num = sizeof(default_kv_table) / sizeof(default_kv_table[0]);

   fdb_kvdb_init(&kvdb, "env", "fdb_kvdb", &default_kv, NULL);
}
```
初始化完成后，程序开始运行，运行过程中加入一个kv。
```c
struct test {
   uint8_t  a;
   uint8_t   b;
};

struct fdb_blob blob;
struct test  p_test;
p_test.a = 1u;
p_test.b = 2u;

fdb_kv_set_blob(&kvdb, "test", fdb_blob_make(&blob, &p_test, sizeof(struct test)));

```
通过以上方式kvdb中增加了test kv，如果在随后升级程序，struct test 改变，会造成test kv读取数据错误。
比如：
```c
struct test {
   uint8_t   a;
   uint8_t   b;
   uint8_t   c;
};

struct fdb_blob blob;
struct test  p_test;

fdb_kv_get_blob(&kvdb, "test", fdb_blob_make(&blob, &p_test, sizeof(struct test)));
```
这原因是不是因为在改变struct test后没有set test kv 直接get test kv造成的？

因为程序的配置信息存储在kvdb中，而调试过程中会对配置结构体的变量增加或减少，就会出现这样的问题。

我是不是应该将test kv添加到default kv中，这样每次升级直接修改kv的ver_num就可以避免这种问题了？

ZakiLiu, 2020-08-26:
想了想，描述的太啰嗦了，修改下。

armink, 2020-08-26:
跟你上面用其他类型是一样的

ZakiLiu, 2020-08-27:
谢谢，朱总！

closed 2020-08-27 by ZakiLiu