armink/FlashDB #33: 关于env 增量升级的逻辑

ZakiLiu, 2020-09-30:
朱总，好！
又打扰您了，想问下env增量升级的逻辑，比如default kv node中有2个结构体，如果结构体中的元素增加或减少后，改变KVDB_VERSION，会更新对于的数据吗？还是说default kv node中有2个结构体，如果再增加一个node后，改变KVDB_VERSION，数据会更新。

我测试修改结构体中的元素，改变KVDB_VERSION，读出的数据不对，所以想请教下env增量升级的逻辑，谢谢您！

armink, 2020-09-30:
没有太看明白你的意思，可以举一下具体的例子吗？两次升级前后 默认 kv 的差异

ZakiLiu, 2020-09-30:
比如：
```c
struct system_info{
    uint32_t                boot_count;
    uint32_t                error_count;
};

struct base_info{
    char    name[10];
    uint32_t  version;
};

static struct system_info  sys_init_info =
{
    .boot_count = 0u,
    .error_count = 0u,
};

tatic struct base_info  base_init_info =
{
    .name = "abc",
    .version = 1,
}

#define KVDB_VERSION    ( 0u )

static struct fdb_default_kv_node system_kv_table[] = {
    {"sys_info", &sys_init_info, sizeof(struct system_info)},
    {"base_info", &base_init_info, sizeof(struct base_info)}
};
```
以上是原始结构体，编译后下载，随后第一种情况是，修改struct system_info结构体，改为如下：
```c
struct system_info{
    uint32_t                boot_count;
    uint32_t                error_count;
    uint8_t                  test_count;
};

static struct system_info  sys_init_info =
{
    .boot_count = 0u,
    .error_count = 0u,
    .test_count  = 1u,
};

#define KVDB_VERSION    ( 1u )
```
修改以上内容，重新下载，我发现读取base_info的name出错了，是不是不能这样做？

第二种情况是，在原来结构体的基础上增加一个结构体：
```c
struct test{
    uint8_t                  test_count;
};

static struct test  test =
{
    .test_count  = 1u,
};

#define KVDB_VERSION    ( 1u )

static struct fdb_default_kv_node system_kv_table[] = {
    {"sys_info", &sys_init_info, sizeof(struct system_info)},
    {"base_info", &base_init_info, sizeof(struct base_info)},
    {"test", &test, sizeof(struct test)}
};
```
是不是需要这样做才能增量升级成功，保证数据不错位？

armink, 2020-09-30:
自动升级没法做到 value 的内容格式发生变化进行升级，主要是 kv 的默认集合发生变化时，自动增加新的 kv

你这种情况，建议：

- 考虑增加新的 kv 
- 设计时留有一定的余量
- 开机检查 system_info 实际存储与固件中 value 大小是否相同，不相同进行手动更新

你觉得这样可以吗

ZakiLiu, 2020-09-30:
可以，朱总，有实现方法就可以。

- 增加新的kv就是我说的第二种方法吧
- 设计留余量，这个目前我的个人能力不足
- 开机检查这个可以，实现比较容易，我先用这个方法

谢谢！

closed 2020-09-30 by ZakiLiu