armink/FlashDB #40: 请问如何可以缩短fdb_kvdb_init()的运行时间

qinhao4832, 2020-10-21:
你好，我的是nrf52832 + SPI flash 的低功耗产品。每次按键开机，实际上就是一次硬复位。fdb_kvdb_init() 实测每次运行大概1秒多时间，按键开机的体验很不顺畅。  请问，修改什么参数 可以缩短fdb_kvdb_init()的运行时间吗？

armink, 2020-10-22:
- kvdb 的容量有多大？大概存储了多少参数？
- 是否也可以试着将 fdb_kvdb_init 放在后台线程中去初始化

qinhao4832, 2020-10-22:
- 容量 256*1024byte.   只存了一个 blob。 大概16byte。
- 这颗ble soc芯片 sdk用的是裸机。
- 另外，fdb_kv_set_blob 返回 FDB_SAVED_FULL 应该怎么处理？目前每次开机这个函数都返回7.   fdb_kvdb_init返回0，是成功的

armink, 2020-10-22:
- KVDB 容量降低一些改为比如 16*4096 试试？
- 如果只有 1 个 blob，返回 FDB_SAVED_FULL  有点奇怪呢，有没有更多日志

qinhao4832, 2020-10-22:
fdb_kv_set_default  (&kvdb);     我尝试调用一次这个以后。  fdb_kv_set_blob 就不再返回 FDB_SAVED_FULL。fdb_kvdb_init 也只消耗50ms左右。目前上述问题，复现不出来了。也许是调试的时候，移植demo的时候 有多个blob，导致的?

armink, 2020-10-22:
有可能的，不可能只有 1 个 blob 在里面，下次可以试试用把所有的 kv print 出来看下