armink/FlashDB #45: kv对校验失败后尝试发现下一个节点时，检索范围的问题

mrsuiqi, 2020-11-24:
函数get_next_kv_addr中，发现pre_kv的crc校验失败，会触发find_next_kv_addr来尝试发现下一个kv_node。
但是在传递参数的的时候终止地址似乎有些问题。
终止地址传递的是sector结束地址减去sector_hdr的长度，但是sector_hdr是放在sector的头部的，这样看起来会导致find的时候没有对sector最后一部分地址进行检查。
劳烦作者看一下我的理解是否有问题

armink, 2020-11-24:
你说的是这里吗？

[image]

mrsuiqi, 2020-11-24:
是的

mrsuiqi, 2020-11-25:
起始地址的话，看了下应该是没什么问题，只是逻辑比较绕。终止地址sector->addr+db_sec_size(db)更合理。

armink, 2020-11-25:
嗯，那你改下真机试试哈，没问题的话可以提交 PR 上来 😄

joechenchen, 2021-01-18:
@mrsuiqi,兄弟，不提交PR吗？

mrsuiqi, 2021-01-19:
最近没有在搞这个了，修复的话比较简单，你那边修复一下就可以了啊

likyo, 2021-01-22:
同样遇到这个问题，最后定位到一个扇区的结尾才会出现此问题，修改后验证OK，线上的还是有bug的，作者还是尽快提交吧

closed 2021-01-23 by armink

joechenchen, 2021-01-23:
看代码起始地址之前加过SECTOR_HDR_DATA_SIZE，我这边是修改成addr = find_next_kv_addr(db, addr , sector->addr + db_sec_size(db))，难道我理解错误了

armink, 2021-01-23:
> 看代码起始地址之前加过SECTOR_HDR_DATA_SIZE，我这边是修改成addr = find_next_kv_addr(db, addr , sector->addr + db_sec_size(db))，难道我理解错误了

多谢反馈，已经改正了

commit 94877499 (2021-01-23): [kvdb] Improve get_next_kv_addr function. Fixed #45