armink/FlashDB #47: TSDB ,SPI FLASH W25Q256,TSDB 首次格式化时间过长，调用fdb_tsl_clean() 耗时过长

HuanDeng, 2020-11-25:
如题，我使用裸机系统移植FlashDB 主要使用了TSDB功能，我的SPIFLASH比较大 32MB，扇区4K共8192个扇区，我发现在第一次初始化的时候似乎FlashDB会擦写每一个扇区，相当的耗时。当存储了一部分数据想要清除数据库记录时，又会重复上述过程。请问有没有办法缩短初始化的格式化时间？

armink, 2020-11-25:
这个是数据库的首次格式化时间，第二次使用就没问题了，好像不太好缩短呢

或者，你确认需要 8192 这么多扇区来存放数据吗？能不能再降低一些

HuanDeng, 2020-12-01:
时序数据库这个功能很好，可以免去自己管理物联网传感器采集的数据，但是时序数据在本机内能够保留得越多越好。另外，是否可以做一张索引表，标记哪些扇区需要擦除，在使用前擦除

armink, 2020-12-01:
这个索引表存哪里呢？写入索引表也需要考虑掉电保护的逻辑，复杂度会不会提高呢。

HikerPan, 2020-12-14:
@HuanDeng 我在之前使用EasyFlash的时候，也遇到类似的情况。
经过一段时间的实践，EasyFlash和FlashDB不太适合使用在过大的内存空间里。
最好的使用场景是对一些kv进行配置，存储空间最好不要超过1M。
太大以后（例如32M），当存储的数据比较多的时候，查找会比较耗时。
我后来的方案就是留512K，给easyflash存储配置数据。其他的数据，使用了littlefs来进行存储。
@armink 老大，不知我的解释是否正确。
我看esp-idf里面也是这样的。NVS其实类似flashdb的功能，存储kv，文件系统存储其他数据。

armink, 2020-12-14:
建议做好数据分区，同一类的数据放在一个数据库中，从数据库的角度去分库，降低检索时间