armink/FlashDB #52: KV数据库，写满如何处理

joechenchen, 2020-12-29:
朱工，你好：
      应用中常常需要保存参数到flash中，TS数据库有回滚的配置，但是KV数据库写满后，会返回FDB_SAVED_FULL，用户应该如何操作，是否需要手动调用fdb_kv_set_default清空数据库，然后重新写入？当前，每个flash扇区4K，配置分区起始地址0，大小为8K。

armink, 2020-12-29:
KV 你都用来存什么数据了？

joechenchen, 2020-12-29:
朱工：
static struct fdb_kvdb kvdb = { 0 };
static uint32_t boot_count = 0;
static time_t boot_time[10] = {0, 1, 2, 3};
static uint8_t test[4000] = {0};
static struct fdb_default_kv_node default_kv_table[] = {
        {"username", "armink", 0}, /* string KV */
        {"password", "123456", 0}, /* string KV */
        {"boot_count", &boot_count, sizeof(boot_count)}, /* int type KV */
        {"boot_time", &boot_time, sizeof(boot_time)},    /* int array type KV */
        {"test", &test, sizeof(test)},    /* 测试写满后的操作*/
};

stErr = fdb_kv_set_blob(kvdb, "test", fdb_blob_make(&blob, &test, sizeof(test)));
用来存储系统参数，想验证下写满后，能否继续写入，于是做了个实验，每次存4000个数据，发现第二次调用fdb_kv_set_blob写入4000个数据时，返回FDB_SAVED_FULL，第三次调用fdb_kv_set_blob又能够正常写入，目前将分区大小增大到12K，连续调用10次fdb_kv_set_blob都没出现FDB_SAVED_FULL。我想咨询下KV操作时，当写满时，是否需要用户去执行fdb_kv_set_default操作。

armink, 2020-12-29:
不需要的，fdb_kv_set_default 相当于格式化，用户的数据都被清空了

joechenchen, 2020-12-29:
你好，也就是说KV数据库写满后，能够自动支持重新从起始地址开始，写入最新参数是吧，如果这样的话，就很方便了。

wwm2021, 2023-04-08:
目前的文件结构有点复杂，计划何时规整？

joechenchen, 2023-04-08:
邮件已收到，谢谢！