armink/FlashDB #56: TSL数据库删除后再次追加遍历后发现部分数据有问题

orange-songqq, 2021-01-04:
根据tsdb_sample.c修改测试程序
1、首先append10条数据，然后遍历一遍数据是没有问题
2、再次重复上述动作
3、调用删除接口，删除数据库中数据【将状态修改为delete】
4、再次重复1、2动作，发现遍历后数据出现问题
代码：
写入数据
` { /* APPEND new TSL (time series log) */
        struct env_status status;

        for (int i = 0; i < 10; i++)
        {
        status.temp = 48 + i;
        status.humi = 100 + i;
        fdb_tsl_append(tsdb, fdb_blob_make(&blob, &status, sizeof(status)));
        FDB_INFO("append the new status.temp (%d) and status.humi (%d)\n", status.temp, status.humi);
        }
    }`
遍历数据库数据
`fdb_tsl_iter(tsdb, query_cb, tsdb);
static bool query_cb(fdb_tsl_t tsl, void *arg)
{
    struct fdb_blob blob;
    struct env_status status;
    fdb_tsdb_t db = arg;

    fdb_blob_read((fdb_db_t) db, fdb_tsl_to_blob(tsl, fdb_blob_make(&blob, &status, sizeof(status))));
    FDB_INFO("[query_cb] queried a TSL: time: %ld, status: %d, temp: %d, humi: %d\n", tsl->time, tsl->status, status.temp, status.humi);

    return false;
}`
删除数据库数据，将状态修改为delete:

`fdb_tsl_clean(tsdb);
  fdb_tsl_iter(tsdb, set_status_delete_cb, tsdb);
  fdb_tsl_iter(tsdb, query_cb, tsdb);`

日志：
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 21, status: 4, temp: 58, humi: 110
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 22, status: 4, temp: 59, humi: 111
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 23, status: 4, temp: 60, humi: 112
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 24, status: 4, temp: 61, humi: 113
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 25, status: 4, temp: 62, humi: 114
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 26, status: 4, temp: 63, humi: 115
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 27, status: 4, temp: 64, humi: 116
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 28, status: 4, temp: 65, humi: 117
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 29, status: 4, temp: 8, humi: 352
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 30, status: 4, temp: 67, humi: 40
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 31, status: 4, temp: 360, humi: 0
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 32, status: 4, temp: 39, humi: 8
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 33, status: 4, temp: 0, humi: 122
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 34, status: 4, temp: 8, humi: 368
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 35, status: 4, temp: 72, humi: 38
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 129, status: 4, temp: 0, humi: -1
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 74, status: 4, temp: 74, humi: 126
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 38, status: 4, temp: 75, humi: 127
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 39, status: 4, temp: 76, humi: 128
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 40, status: 4, temp: 77, humi: 129
[FlashDB][sample][tsdb] ===========================================================
>

orange-songqq, 2021-01-04:
不好意思，没写完就发出去了。。。

这两行数据是有问题：与设置进去的数据不符，time的返回值也是有问题的
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 129, status: 4, temp: 0, humi: -1
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 74, status: 4, temp: 74, humi: 126

还有个问题就是，调用删除接口之后，调用了fdb_tsl_query_count接口查询FDB_TSL_DELETED与FDB_TSL_USER_STATUS1的count，发现返回值为1152
调用fdb_tsl_iter接口查看数据库中的数据，打印出来的数据（16条）比append数据（20条）少4个

armink, 2021-01-04:
你好，根据你的描述，简单写了下测试代码，多次尝试后没有复现问题，确认下面的测试代码是否可以在你的平台复现问题

```C
static bool query_cb(fdb_tsl_t tsl, void *arg)
{
    struct fdb_blob blob;
    struct env_status status;
    fdb_tsdb_t db = arg;

    fdb_blob_read((fdb_db_t) db, fdb_tsl_to_blob(tsl, fdb_blob_make(&blob, &status, sizeof(status))));
    FDB_INFO("[query_cb] queried a TSL: time: %ld, status: %d, temp: %d, humi: %d\n", tsl->time, tsl->status, status.temp, status.humi);

    return false;
}

static bool set_status_delete_cb(fdb_tsl_t tsl, void *arg)
{
    fdb_tsdb_t db = arg;

    FDB_INFO("set the TSL (time %ld) status from %d to %d\n", tsl->time, tsl->status, FDB_TSL_DELETED);
    fdb_tsl_set_status(db, tsl, FDB_TSL_DELETED);

    return false;
}

void github_issue56_test(void)
{
    struct fdb_blob blob;

    { /* APPEND new TSL (time series log) */
        struct env_status status;

        for (int i = 0; i < 10; i++)
        {
            status.temp = 48 + i;
            status.humi = 100 + i;
            fdb_tsl_append(tsdb, fdb_blob_make(&blob, &status, sizeof(status)));
            FDB_INFO("append the new status.temp (%d) and status.humi (%d)\n", status.temp, status.humi);
        }
        fdb_tsl_iter(tsdb, query_cb, tsdb);
    }

    { /* APPEND new TSL (time series log) */
        struct env_status status;

        for (int i = 0; i < 10; i++)
        {
            status.temp = 48 + i;
            status.humi = 100 + i;
            fdb_tsl_append(tsdb, fdb_blob_make(&blob, &status, sizeof(status)));
            FDB_INFO("append the new status.temp (%d) and status.humi (%d)\n", status.temp, status.humi);
        }
        fdb_tsl_iter(tsdb, query_cb, tsdb);
    }

    fdb_tsl_clean(tsdb);
    fdb_tsl_iter(tsdb, set_status_delete_cb, tsdb);
    fdb_tsl_iter(tsdb, query_cb, tsdb);

    { /* APPEND new TSL (time series log) */
        struct env_status status;

        for (int i = 0; i < 10; i++)
        {
            status.temp = 48 + i;
            status.humi = 100 + i;
            fdb_tsl_append(tsdb, fdb_blob_make(&blob, &status, sizeof(status)));
            FDB_INFO("append the new status.temp (%d) and status.humi (%d)\n", status.temp, status.humi);
        }
        fdb_tsl_iter(tsdb, query_cb, tsdb);
[... 57 lines cut ...]
[FlashDB][issue] [query_cb] queried a TSL: time: 17, status: 2, temp: 55, humi: 107
[FlashDB][issue] [query_cb] queried a TSL: time: 18, status: 2, temp: 56, humi: 108
[FlashDB][issue] [query_cb] queried a TSL: time: 19, status: 2, temp: 57, humi: 109
[FlashDB][tsl][log] All sector format finished.
[FlashDB][issue] append the new status.temp (48) and status.humi (100)
[FlashDB][issue] append the new status.temp (49) and status.humi (101)
[FlashDB][issue] append the new status.temp (50) and status.humi (102)
[FlashDB][issue] append the new status.temp (51) and status.humi (103)
[FlashDB][issue] append the new status.temp (52) and status.humi (104)
[FlashDB][issue] append the new status.temp (53) and status.humi (105)
[FlashDB][issue] append the new status.temp (54) and status.humi (106)
[FlashDB][issue] append the new status.temp (55) and status.humi (107)
[FlashDB][issue] append the new status.temp (56) and status.humi (108)
[FlashDB][issue] append the new status.temp (57) and status.humi (109)
[FlashDB][issue] [query_cb] queried a TSL: time: 20, status: 2, temp: 48, humi: 100
[FlashDB][issue] [query_cb] queried a TSL: time: 21, status: 2, temp: 49, humi: 101
[FlashDB][issue] [query_cb] queried a TSL: time: 22, status: 2, temp: 50, humi: 102
[FlashDB][issue] [query_cb] queried a TSL: time: 23, status: 2, temp: 51, humi: 103
[FlashDB][issue] [query_cb] queried a TSL: time: 24, status: 2, temp: 52, humi: 104
[FlashDB][issue] [query_cb] queried a TSL: time: 25, status: 2, temp: 53, humi: 105
[FlashDB][issue] [query_cb] queried a TSL: time: 26, status: 2, temp: 54, humi: 106
[FlashDB][issue] [query_cb] queried a TSL: time: 27, status: 2, temp: 55, humi: 107
[FlashDB][issue] [query_cb] queried a TSL: time: 28, status: 2, temp: 56, humi: 108
[FlashDB][issue] [query_cb] queried a TSL: time: 29, status: 2, temp: 57, humi: 109
```

orange-songqq, 2021-01-05:
你好，我之前的测试代码和你提供的也没有太大差异，现在我将你的测试代码拷贝到我这边的测试环境上
1、首先你这边提供的日志和你的测试代码有些许不同吧，在删除修改状态后你的测试代码是有遍历数据库的，但是日志中没有这段日志，我的测试日志中是有该日志打印的，循环多次，且与我上面描述的一样，【打印出来的数据（16条）比append数据（20条）少4个】
2、我这边在修改完状态后，再次追加的数据的状态是4不是2
3、可能需要麻烦你在目前的基础上再追加10条数据，然后看一下是否数据有不同的地方
我这边测试直接追加10条数据是没有问题，但是日志有循环打印的情况
再追加10条就会出现数据错乱的问题
4、将状态改为delete，这些数据并不能被再次加入的数据覆盖掉，为什么

下面是用你提供的测试程序测试的日志，我这边的日志再调用clean接口后就会出现不断的循环日志，所以我就放部分日志：

> [FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 1, status: 2, temp: 48, humi: 100
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 2, status: 2, temp: 49, humi: 101
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 3, status: 2, temp: 50, humi: 102
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 4, status: 2, temp: 51, humi: 103
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 5, status: 2, temp: 52, humi: 104
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 6, status: 2, temp: 53, humi: 105
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 7, status: 2, temp: 54, humi: 106
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 8, status: 2, temp: 55, humi: 107
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 9, status: 2, temp: 56, humi: 108
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 10, status: 2, temp: 57, humi: 109
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 11, status: 2, temp: 48, humi: 100
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 12, status: 2, temp: 49, humi: 101
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 13, status: 2, temp: 50, humi: 102
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 14, status: 2, temp: 51, humi: 103
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 15, status: 2, temp: 52, humi: 104
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 16, status: 2, temp: 53, humi: 105
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 17, status: 2, temp: 54, humi: 106
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 18, status: 2, temp: 55, humi: 107
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 19, status: 2, temp: 56, humi: 108
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 20, status: 2, temp: 57, humi: 109
[FlashDB][tsl][log] All sector format finished.
[FlashDB][sample][tsdb] set the TSL (time 1) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 2) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 3) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 4) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 5) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 6) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 7) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 8) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 9) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 10) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 11) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 12) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 13) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 14) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 15) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 16) status from 2 to 4
[FlashDB][sample][tsdb] set the TSL (time 1) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 2) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 3) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 4) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 5) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 6) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 7) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 8) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 9) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 10) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 11) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 12) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 13) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 14) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 15) status from 4 to 4
[FlashDB][sample][tsdb] set the TSL (time 16) status from 4 to 4
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 1, status: 4, temp: 48, humi: 100
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 2, status: 4, temp: 49, humi: 101
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 3, status: 4, temp: 50, humi: 102
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 4, status: 4, temp: 51, humi: 103
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 5, status: 4, temp: 52, humi: 104
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 6, status: 4, temp: 53, humi: 105
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 7, status: 4, temp: 54, humi: 106
[... 85 lines cut ...]
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 32, status: 4, temp: 39, humi: 8
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 33, status: 4, temp: 0, humi: 102
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 34, status: 4, temp: 8, humi: 368
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 35, status: 4, temp: 52, humi: 38
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 109, status: 4, temp: 16711680, humi: -1
[... cut ...]
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 35, status: 4, temp: 52, humi: 38
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 109, status: 4, temp: 16711680, humi: -1
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 54, status: 4, temp: 54, humi: 106
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 38, status: 4, temp: 55, humi: 107
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 39, status: 4, temp: 56, humi: 108
[FlashDB][sample][tsdb] [query_cb] queried a TSL: time: 40, status: 4, temp: 57, humi: 109

armink, 2021-01-05:
感觉你的 Flash 驱动问题概率会比较高，你用的是什么平台？有用 demo 里面的演示平台验证一下吗？

orange-songqq, 2021-01-19:
是我们自己移植的问题，谢谢~~

orange-songqq, 2021-01-19:
已解决

closed 2021-01-19 by orange-songqq