armink/FlashDB #59: FlashDB的源程序中，为什么很多语句不判断下一层函数的返回数值，而是直接返回？

xql923, 2021-01-19:
朱先生你好，感谢您的FlashDB！

接上一条问题继续，我找到问题的根本原因，在于我的驱动中读取延时时间太短导致读取数据不全而出现读取错误。

但是我在排查故障的过程中发现FlashDB的实现程序中，从上层调用下一层函数的过程中，很多不判断函数返回值，而是直接返回，比如：
1、fdb_kvdb.c源文件get_kv函数中，第588行_fdb_flash_read((fdb_db_t)db, kv.addr.value, (uint32_t *) value_buf, read_len)语句，没有判断是否读取正确，而是直接返回。

2、fdb_utils.c源文件中_fdb_flash_read函数中，第256行fal_partition_read(db->storage.part, addr, (uint8_t *) buf, size)也没有判断返回，而是直接返回FDB_NO_ERR。

3、fdb_kvdb.c文件中，第530行find_kv(fdb_kvdb_t db, const char *key, fdb_kv_t kv)函数中调用read_kv(db, kv)也没有判断返回，而是直接返回true。

4、例程中FlashDB移植的read(long offset, uint8_t *buf, size_t size)函数，也没有判断返回数值，而是直接返回四则。

       还有很多处，没有判断函数执行返回数值。
       这样带来的问题是，当底层一个函数执行出现错误时，但是仍然是按照正确时返回，最终会出现读取或者写入的数据是错误的，而且程序健壮性不够强，排查故障很不方便。

      朱先生是嵌入式行业大牛，请问，您这么做的原因是什么？有什么特殊考虑吗？

xql923, 2021-01-19:
发现，源程序中所有使用__fdb_flash_read(fdb_db_t db, uint32_t addr, void *buf, size_t size)函数都没有判断返回数值，判断是否执行正确，全部按照正确的处理。

xql923, 2021-01-19:
主要修改如下：

mysterywolf, 2021-01-19:
@armink  我和@xql923交流过了，他是在实际工程中遇到的这个问题。由于FlashDB底层没有充分判断错误代码，导致上层软件误认为是正确，进而引起一系列的连锁反应，返回值/错误代码 的检查望重视（包括RTT内核/组件以及其他重要软件包）。

armink, 2021-01-19:
可能当时考虑的不是很周到，欢迎提交 PR 哈