armink/FlashDB #60: #if (KV_STATUS_TABLE_SIZE >= FDB_DIRTY_STATUS_TABLE_SIZE) may not work as expected

ivan7wl, 2021-01-20:
In `fdb_kvdb.c::del_kv()` function, `FDB_SECTOR_STORE_STATUS_NUM` and `FDB_SECTOR_DIRTY_STATUS_NUM` is an enum constant, if used in #if statement, the preprocessor will assume they are undefined macros, and evaluate to 0, so the #if statement will always be true.
Suggest use:
```
uint8_t status_table[KV_STATUS_TABLE_SIZE >= FDB_DIRTY_STATUS_TABLE_SIZE ? KV_STATUS_TABLE_SIZE : DIRTY_STATUS_TABLE_SIZE];
```

在`fdb_kvdb.c::del_kv()`函数中，`FDB_SECTOR_STORE_STATUS_NUM`和`FDB_SECTOR_DIRTY_STATUS_NUM`是枚举常量，在#if条件编译语句中，预处理器将它们看作是未定义的宏，求值为0，#if语句永远为真。
建议使用：
```
uint8_t status_table[KV_STATUS_TABLE_SIZE >= FDB_DIRTY_STATUS_TABLE_SIZE ? KV_STATUS_TABLE_SIZE : DIRTY_STATUS_TABLE_SIZE];
```

AdrianGuo, 2021-01-20:
咋不提价PR呢

armink, 2021-01-20:
两种方式都可以的，各有利弊，还可定义一个 MAX 宏来进一步简化

```
uint8_t status_table[KV_STATUS_TABLE_SIZE >= FDB_DIRTY_STATUS_TABLE_SIZE ? KV_STATUS_TABLE_SIZE : DIRTY_STATUS_TABLE_SIZE];
```

```C
#if (KV_STATUS_TABLE_SIZE >= FDB_DIRTY_STATUS_TABLE_SIZE)
    uint8_t status_table[KV_STATUS_TABLE_SIZE];
#else
    uint8_t status_table[DIRTY_STATUS_TABLE_SIZE];
#endif
```

ivan7wl, 2021-01-20:
> 两种方式都可以的，各有利弊，还可定义一个 MAX 宏来进一步简化
>
> ```
> uint8_t status_table[KV_STATUS_TABLE_SIZE >= FDB_DIRTY_STATUS_TABLE_SIZE ? KV_STATUS_TABLE_SIZE : DIRTY_STATUS_TABLE_SIZE];
> ```
>
> ```c
> #if (KV_STATUS_TABLE_SIZE [...]

用#if条件编译，实际求值的是
```
#if FDB_STATUS_TABLE_SIZE(0) >= FDB_STATUS_TABLE_SIZE(0)
```
而不是预期的
```
#if FDB_STATUS_TABLE_SIZE(6) >= FDB_STATUS_TABLE_SIZE(4)
```

armink, 2021-01-21:
其实我的预期也不是 

```
#if FDB_STATUS_TABLE_SIZE(6) >= FDB_STATUS_TABLE_SIZE(4)
```

应该会根据 Flash 的写粒度，自动计算出实际占用的字节数，比如：当写粒度为 1 时

[image]

ivan7wl, 2021-01-23:
> 其实我的预期也不是
> 
> ```
> #if FDB_STATUS_TABLE_SIZE(6) >= FDB_STATUS_TABLE_SIZE(4)
> ```
> 
> 应该会根据 Flash 的写粒度，自动计算出实际占用的字节数，比如：当写粒度为 1 时
> 
> [image]

问题是预处理器只认识宏定义啊，用-Wundef参数编译一下就看到了

armink, 2021-01-23:
👍 感谢你的反馈哈，已修正

commit 218c4df0 (2021-01-23): [kvdb] Improve del_kv code. Fixed #60
closed 2021-01-23 by armink