armink/FlashDB #61: 改进建议：可重入的接口

ivan7wl, 2021-01-20:
场景：多核处理器，有多个kvdb实例和多个cpu，每个cpu独占使用自己的kvdb实例。
目前kvdb的`set_kv()`，`del_kv()`等函数使用了static变量，无法支持重入，需要在调用时加全局锁。
如果暂不打算支持，建议在api文档中说明哪些接口无法重入。

armink, 2021-01-20:
可以用，blob KV  API，如 fdb_kv_set_blob     API

ivan7wl, 2021-01-20:
> 可以用，blob KV API，如 fdb_kv_set_blob API

fdb_kv_set_blob调用了set_kv，set_kv里面使用了static变量kv和sector，就是不可重入的。

armink, 2021-01-20:
fdb_kv_set_blob 里有 lock

ivan7wl, 2021-01-21:
> fdb_kv_set_blob 里有 lock

lock的设计意图是避免多个进程访问同一个kvdb实例产生冲突，但是现在不同的进程访问不同的kvdb实例，仍可能在static变量上产生冲突，所以即使是不同的实例，也要加全局锁。

armink, 2021-01-21:
明白你的意思里，可以移除掉 set_kv/del_kv 中的 static 修饰，方便提交一个 PR 改进下吗

ivan7wl, 2021-01-22:
我这边修改比较简单，就直接说了吧：
1. del_kv()里用到的last_is_complete_del，把last_is_complete_del放在struct fdb_tsdb结构体里，通过db->last_is_complete_del引用；
2. set_kv()里的kv和sector变量，去掉static修饰，直接定义在栈上，会稍微多消耗一些栈空间

armink, 2021-01-22:
> 我这边修改比较简单，就直接说了吧：
> 
> 1. del_kv()里用到的last_is_complete_del，把last_is_complete_del放在struct fdb_tsdb结构体里，通过db->last_is_complete_del引用；
> 2. set_kv()里的kv和sector变量，去掉static修饰，直接定义在栈上，会稍微多消耗一些栈空间

完全可以的，可以尝试提交一下 PR 吗

likyo, 2021-01-25:
fdb_kv_get（）这个函数不可重入，并且没法读取汉字字符串。因为加了ascii判断

armink, 2021-02-02:
fixed in b3041a177097ec0343d53565c6bef0be4377bd60

closed 2021-02-02 by armink