armink/FlashDB #63: FDB KV  如何配置存储的最大字符数

hcb900330, 2021-01-31:
FDB_STR_KV_VALUE_MAX_SIZE                1024*100   为获取的最大数  存储的最大数在哪里配置

armink, 2021-01-31:
在 fdb_cfg.h 中配置即可

hcb900330, 2021-01-31:
#define FDB_WRITE_GRAN                 8  这个带表啥  意思  8*4k么？

hcb900330, 2021-01-31:
#define FDB_WRITE_GRAN                 8*4  改成这样数据大以后 还是存不进去

hcb900330, 2021-01-31:
#define FDB_WRITE_GRAN                1  这个确实配置错了
但是该正确以后，value字符数多了  还是存不进去  不知道这value的最大数  在哪里配置

hcb900330, 2021-01-31:
> #define FDB_WRITE_GRAN 1 这个确实配置错了
> 但是该正确以后，value字符数多了 还是存不进去 不知道这value的最大数 在哪里配置

好像没有这个参数配置

hcb900330, 2021-01-31:
> 在 fdb_cfg.h 中配置即可

kv大小不好设置   我想换ts试试的     result = fdb_tsdb_init(&tsdb, "liandong", "fdb_tsdb1", get_time, 20*1024, NULL);
还是不能初始化
[FlashDB][tsl][liandong] (applications\FlashDB\src\fdb_tsdb.c:748) (max_len < db_sec_size(db)) has assert failed at fdb_tsdb_init.

armink, 2021-02-01:
如果是设置 TSDB 的 MAX_LEN ，可以通过 fdb_tsdb_init 来设置

hcb900330, 2021-02-01:
> 如果是设置 TSDB 的 MAX_LEN ，可以通过 fdb_tsdb_init 来设置
我设置了20*1024 都设置不了的
[FlashDB][tsl][liandong] (applications\FlashDB\src\fdb_tsdb.c:748) (max_len < db_sec_size(db)) has assert failed at fdb_tsdb_init.

armink, 2021-02-01:
那说明是你的 扇区设置的太小了，可以通过 control 来调整扇区大小 http://armink.gitee.io/flashdb/#/zh-cn/api?id=%e6%8e%a7%e5%88%b6-tsdb

hcb900330, 2021-02-01:
> 那说明是你的 扇区设置的太小了，可以通过 control 来调整扇区大小 http://armink.gitee.io/flashdb/#/zh-cn/api?id=%e6%8e%a7%e5%88%b6-tsdb

flashdb  kv应该通过什么配置来设置大小

hcb900330, 2021-02-01:
> 那说明是你的 扇区设置的太小了，可以通过 control 来调整扇区大小 http://armink.gitee.io/flashdb/#/zh-cn/api?id=%e6%8e%a7%e5%88%b6-tsdb

修改了下，还是没有用，最大只能设置1024

                          int len=1024*1024;  
			  int len2=1024*50; 
			  fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_ROLLOVER, &len);
			  fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_SEC_SIZE, &len2);
			  rt_thread_delay(100);
        result = fdb_tsdb_init(&tsdb, "liandong", "fdb_tsdb1", get_time, 40*1024, NULL);
        if (result != FDB_NO_ERR) {
            return -1;
        }
[FlashDB] You can get the latest version on https://github.com/armink/FlashDB .
[FlashDB] (applications\FlashDB\src\fdb.c:68) (db->sec_size % block_size == 0) has assert failed at _fdb_init_ex.