armink/FlashDB #66: _fdb_write_status()函数执行前是否先擦除再写入？

mysterywolf, 2021-02-28:
由于最近github间歇性被墙，我代 @xql923 提问：

硬件环境：
       MCU：STM32F413；SPI Flash：W25Q256
      
      配置FlashDB参数：FDB_WRITE_GRAN == 1
   
       W25Q256每次按照一个扇区擦除4096字节，一个扇区只有在擦除后才能再次写入，如果扇区内数据没有擦除而直接写入，会写入失败。

问题描述：
       移植FlashDB保存参数和数据，但是在实际仿真调试时，fdb_utils.c文件中_fdb_write_status()函数总是执行失败，仿真发现，_fdb_write_status()函数每次值写入一个字节，写入前FlashDB并没有考虑擦除所在扇区数据，而是直接写入，这样会导致写入失败。

       请问，_fdb_write_status()函数没有考虑在写入前，擦除相应扇区的数据吗？
       是否需要在移植接口函数中特殊考虑？

armink, 2021-02-28:
理论上不存在这种情况，默认都是上电会检测 Flash 是否有做过格式化，必要时，系统会自动对 Flash 完成格式化动作

格式化时就会对 Flash 执行擦除动作

xql923, 2021-02-28:
朱经理你好，_fdb_write_status()函数这个函数就只写一个字节，我在写之前先读取位置所在扇区的全部数据，发现其并没有格式化。

     请问，这个函数，你程序中在写入之前有针对性考虑格式化吗？，

armink, 2021-02-28:
手动 fdb_kv_set_default 试一下？

xql923, 2021-02-28:
我初始化多次，都是这样的结果。

我是这样测试的，在写入之前，首先读取，查看地址所在扇区的数据是否全部是0xFF，如果有一个不等于0xFF，就说明扇区没有擦除，写入肯定失败。

xql923, 2021-02-28:
fdb_utils.c文件中，请问：
fdb_err_t _fdb_write_status(fdb_db_t db, uint32_t addr, uint8_t status_table[], size_t status_num, size_t status_index)
{

#if (FDB_WRITE_GRAN == 1)
   
    --------------//这个函数在运行之前，程序中，由针对addr + byte_index这个地址所在扇区进行格式化了吗？

    result = _fdb_flash_write(db, addr + byte_index, (uint32_t *)&amp;status_table[byte_index], 1);

........
}

xql923, 2021-03-01:
朱经理你好，在fdb_utils.c文件中，请问：
fdb_err_t _fdb_write_status(fdb_db_t db, uint32_t addr, uint8_t status_table[], size_t status_num, size_t status_index)
{
......

           #if (FDB_WRITE_GRAN == 1)

           -------//这个函数在运行之前，程序中，由针对addr + byte_index这个地址所在扇区进行格式化了吗？
           result = _fdb_flash_write(db, addr + byte_index, (uint32_t *)&amp;status_table[byte_index], 1);

         #endif

........
}

armink, 2021-03-01:
写之前肯定是有擦除，否则也没法写入的

xql923, 2021-03-01:
朱先生你好，谢谢你的回复！

请问，擦除这个动作是你在程序中已经做好了，还是需要在接口函数中，我来考虑擦除？

armink, 2021-03-01:
用户不需要考虑擦写的时机，只需提供好移植函数就好，建议详细参考下 demo 文件下的示例，并在真机上试用

xql923, 2021-03-01:
收到，谢谢朱先生！

现在Github打不开，没有办法下载。

armink, 2021-03-01:
gitee 上也有，可以搜一下

xql923, 2021-03-02:
朱先生你好，我的SPI Flash并没有采用SFUD，而是FlashDB的fal port接口直接以扇区读、写和删除nor Flash，请问：

xql923, 2021-03-02:
使用的是SPI Flash：W25Q256，每个扇区的大小为：4096字节

xql923, 2021-03-03:
结帖。
将程序实现看了一遍，学习了朱先生的设计精巧之处。特别是针对Nor Flash的写入特点，巧妙的以bit设置状态位实现KV状态的设置，进而实现了所谓“掉电恢复”的功能。

xql923, 2021-03-03:
“磨损均衡”的实现也很巧妙。

closed 2021-03-03 by mysterywolf