armink/FlashDB #68: sector_iterator()调用时传入了未初始化的sector参数

ivan7wl, 2021-03-06:
review代码发现某些变量未初始化就被使用了，以kvdb的gc_collect()函数为例：

1. gc_collect()在栈上定义了`struct kvdb_sec_info sector;`，此时sector结构体未初始化，各个字段是随机值
2. sector_iterator()中，首先调用`read_sector_info(db, sec_addr, sector, false);`，但是如果magic检查失败，没有给`sector->status.store`赋值，就直接return了
3. sector_iterator()接下来会引用`sector->status.store`做判断，但此时该字段可能是未初始化的随机值

请作者分析评估一下是否会造成问题，以及其他代码中是否有类似问题。

armink, 2021-03-06:
保证 gc 操作的地方对 sector->check_ok 有检查就好，这个值是每次 read_sector_info 都会有结果。

ivan7wl, 2021-03-06:
其他的地方也有这种情况，不知道有没有风险？最好还是全面检查一下。这种未初始化的变量可能引入安全弱点，也会导致代码调试维护比较费力。

armink, 2021-03-06:
那对这种情况，感觉这样修改是不是就行了

```c
struct kvdb_sec_info sector = { .check_ok = false };
```

ivan7wl, 2021-03-06:
> 那对这种情况，感觉这样修改是不是就行了
> 
> ```c
> struct kvdb_sec_info sector = { .check_ok = false };
> ```

可以的，但是最好检查逻辑部分是否有风险。