armink/FlashDB #69: kv值长度大于sector长度，fdb_kv_set_blob返回FDB_SAVED_FULL

youjbl, 2021-03-08:
朱工，
您好！

我在使用kvdb时，创建的kv表值长度为4473，发现在fdb_kv_set_blob写入数据库时，一直返回FDB_SAVED_FULL。我使用的norflash型号为W25Q64JV，sector大小为4096。跟踪调试发现，在sector_iterator函数中，执行alloc_kv_cb回调时，
static bool alloc_kv_cb(kv_sec_info_t sector, void *arg1, void *arg2)
{
    struct alloc_kv_cb_args *arg = arg1;

    /* 1. sector has space
     * 2. the NO dirty sector
     * 3. the dirty sector only when the gc_request is false */
    if (sector->check_ok && sector->remain > arg->kv_size
            && ((sector->status.dirty == FDB_SECTOR_DIRTY_FALSE)
                    || (sector->status.dirty == FDB_SECTOR_DIRTY_TRUE && !arg->db->gc_request))) {
        *(arg->empty_kv) = sector->empty_kv;
        return true;
    }

    return false;
}
条件 sector->remain > arg->kv_size一直不能满足。

请问kv值大小不能大于sector大小吗？ 如果是，该如何处理呢？

armink, 2021-03-09:
可以通过 control API 加大一下扇区大小，比如：到16KB，就行了，详见 http://armink.gitee.io/flashdb/#/zh-cn/api?id=%e6%8e%a7%e5%88%b6-kvdb

youjbl, 2021-03-09:
朱工，感谢您的回复！ 我后来也发现了文档上有写。

另外，再请教一个问题：

我在测试KVDB连续写入耗时情况，发现调用fdb_kv_set_blob写入数据库，会随着写入数据量增多，写入时间会越长。
测试条件：
1. W25Q64JV QSPI 4线；
2. 分区大小为2M；
3. 采用了一个结构体（大小为837Bytes）作为kv值连续写入数据库直至数据库写满。 

测试结果：写入耗时时间从最开始的40ms左右，到最后数据库写满，写入耗时时间增加至800ms。写满后，会耗费10.8s时间数据库擦除操作，然后写入耗时时间又从40ms开始递增，循环。

跟踪调试发现，在alloc_kv中，sector_iterator耗费时间非常长，耗时722ms。(注：下图为数据库快写满时，截获的运行时间统计）

请问，这个问题可以优化吗？

感谢！

| |
hongjiang
|
|
lsqwjianghong@163.com
|
签名由网易邮箱大师定制

armink, 2021-04-01:
可以减少分区大小试试，比较你的存储容量不需要那么大