armink/FlashDB #84: 多个时序数据库如何初始化?

leven99, 2021-07-28:
`#define NOR_FLASH_DEV_NAME     "norflash0"`
`#define NOR_FLASH_PART0_NAME   "faultrecord"`
`#define NOR_FLASH_PART1_NAME   "chargerecord"`
`#define NOR_FLASH_PART2_NAME   "reserve"`
`#define NOR_FLASH_PART3_NAME   "download"`

`#define FAL_FLASH_DEV_TABLE   \ `
`{                                                     \`
`    &nor_flash0,                              \`
`}`

`#define FAL_PART_TABLE  \`
`{                                         \`
`    {FAL_PART_MAGIC_WORD, NOR_FLASH_PART0_NAME, NOR_FLASH_DEV_NAME,             0, 1024*1024*2, 0},           \`
`    {FAL_PART_MAGIC_WORD, NOR_FLASH_PART1_NAME, NOR_FLASH_DEV_NAME, 1024*1024*2*1, 1024*1024*2, 0}, \`
 `   {FAL_PART_MAGIC_WORD, NOR_FLASH_PART2_NAME, NOR_FLASH_DEV_NAME, 1024*1024*2*2, 1024*1024*2, 0}, \`
 `   {FAL_PART_MAGIC_WORD, NOR_FLASH_PART3_NAME, NOR_FLASH_DEV_NAME, 1024*1024*2*3, 1024*1024*2, 0}, \`
`}`

## 初始化两种不同数据结构的时序数据库时如何操作呢？

是否可以以如下方式连续初始化两个数据库：

`fdb_tsdb_init(&s_faultdb,     "faultdb",    NOR_FLASH_PART0_NAME, get_time, 128, NULL);`
`fdb_tsdb_init(&s_chargedb, "chargedb", NOR_FLASH_PART1_NAME, get_time, 512, NULL);`

armink, 2021-07-28:
用法没问题的

sudashannon, 2022-04-06:
> 用法没问题的

我看了下fdb_kvdb_init这个接口设计，没看到有支持可重入的啊，里面对fal初始化两次岂不是会有问题？