armink/FlashDB #90: How to install Flash DB in ESP32-S2 wrover board using SPI Flash memory

kalpanavengala, 2021-08-10:
Hi,
How to use flash DB in ESP32 and how to install Flash DB library into ESP32-S2-wrover sola board using SPi flash memory.Csn you please provide me any example programs using time series and Key value storage

armink, 2021-08-11:
You can using the ESP8266 demo for test. https://github.com/armink/FlashDB/tree/master/demos/esp8266_spi_flash

kalpanavengala, 2021-08-12:
Hi Armink,
Thanks for the reply.
I am using esp32-s2 wrover module, esp-idf 4.2.1 version and using the
Vscode tool .I have run the code and am getting error while building the
code. Please give me your opinion on this.
I have included the flashdb.h header file in the components/FlashDB--> *inc*
folder. how to include the flashdb library files in the project folder
without getting errors.
for your reference
here run the command idf.py build and not able to run the make command for
esp32s2
.

*./main/hello_world_main.c:16:10: fatal error: flashdb.h: No such file or
directory #include "flashdb.h"          ^~~~~~~~~~~compilation
terminated.[916/928] Building CXX object
esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.objninja:
build stopped: subcommand failed.ninja failed with exit code 1*

kalpanavengala, 2021-08-12:
Hi Armink,

I am able to flash the code now after setting the Nor flash address as
0x3F000000 by referring to the esp32s2 tech manual.
but I am getting the partition write error. Please refer to the log file
and give me your suggestion to resolve the issue and to perform read and
write operations.

Please see the below log data for your reference:

Hello world!
This is ESP32 chip with 1 CPU cores, WiFi, silicon revision 0, 4MB external
flash
[I/FAL] Flash Abstraction Layer (V0.5.0) initialize success.
[FlashDB][kv][env] (../main/fdb_kvdb.c:1633) KVDB size is 16384 bytes.
[FlashDB][kv][env] (../main/fdb_kvdb.c:1437) Sector header info is
incorrect. Auto format this sector (0x00000000).
[FlashDB][kv][env] (../main/fdb_kvdb.c:1437) Sector header info is
incorrect. Auto format this sector (0x00001000).
[FlashDB][kv][env] (../main/fdb_kvdb.c:1437) Sector header info is
incorrect. Auto format this sector (0x00002000).
[FlashDB][kv][env] (../main/fdb_kvdb.c:1437) Sector header info is
incorrect. Auto format this sector (0x00003000).
[FlashDB][kv][env] All sector header is incorrect. Set it to default.
[FlashDB] FlashDB V1.0.99 is initialize success.
[FlashDB] You can get the latest version on
https://github.com/armink/FlashDB .
[FlashDB][sample][kvdb][basic] ==================== kvdb_basic_sample
====================
[FlashDB][sample][kvdb][basic] get the 'boot_count' failed
[FlashDB][sample][kvdb][basic] set the 'boot_count' value to 1
[FlashDB][sample][kvdb][basic]
===========================================================
[FlashDB][sample][kvdb][string] ====================
kvdb_type_string_sample ====================
[FlashDB][sample][kvdb][string] create the 'temp' string KV, value is: 36C
[FlashDB][sample][kvdb][string] set 'temp' value to 38C
[FlashDB][kv][env] (../main/fdb_kvdb.c:866) Not found 'temp' in KV.
[FlashDB][sample][kvdb][string] delete the 'temp' finish
[FlashDB][sample][kvdb][string]
===========================================================
[FlashDB][sample][kvdb][blob] ==================== kvdb_type_blob_sample
====================
[FlashDB][sample][kvdb][blob] create the 'temp' blob KV, value is: 36
[FlashDB][sample][kvdb][blob] set 'temp' value to 38
[FlashDB][kv][env] (../main/fdb_kvdb.c:866) Not found 'temp' in KV.
[FlashDB][sample][kvdb][blob] delete the 'temp' finish
[FlashDB][sample][kvdb][blob]
===========================================================
[FlashDB][tsl][log] Sector (0x00000000) header info is incorrect.
[FlashDB][tsl][log] All sector format finished.
[FlashDB][tsl][log] (../main/fdb_tsdb.c:783) TSDB (log) oldest sectors is
0x00000000, current using sector is 0x00000000.
[FlashDB][sample][tsdb] ==================== tsdb_sample
====================
[FlashDB][sample][tsdb] append the new status.temp (36) and status.humi (85)
*[E/FAL] (fal_partition_write:424) Partition write error! Partition address
out of bound.*
[FlashDB][sample][tsdb] append the new status.temp (38) and status.humi (90)
[FlashDB][sample][tsdb] query count is: 0
[FlashDB][sample][tsdb]
===========================================================
Restarting in 1000 seconds...
Restarting in 999 seconds...
Restarting in 998 seconds...
Restarting in 997 seconds...
Restarting in 996 seconds.

Kindly give me the reply and your suggestions/ any procedure to resolve
this issue.

Thanks,
Kalpana

armink, 2021-08-13:
@syj0925 Can you give some suggestions?

kaans, 2022-04-25:
It looks like the address to the partition for FlashDB is not correct and the program tries to write to an invalid memory region:
`*[E/FAL] (fal_partition_write:424) Partition write error! Partition address
out of bound.*`

I pushed an example for ESP32 that uses the esp_partition_* api which resolves the correct addresses for the partition automatically: https://github.com/armink/FlashDB/tree/master/demos/esp32_spi_flash