armink/FlashDB #100: FDB_TSL_DELETED: how to delete or rotate?

Grabber, 2021-09-07:
@armink

Thank you so much for FlashDB, lovely piece of C code!

I was doing some experiments today and have one question about **tsdb** and deletion. Could you please help me?

From docs I understood that the appended blobs are never deleted, but can be marked (by the user) with `FDB_TSL_DELETED` flag for post-iterator filtering.

```
bool file_mode = true;
uint32_t sec_size = 4096, db_size = sec_size * 4;

fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_SEC_SIZE, &sec_size);
fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_MAX_SIZE, &db_size);
fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_FILE_MODE, &file_mode);
```

What happens when the `tsdb` reaches the `sec_size * 4` upper size limit? Will it delete older segments or rotate? How do I know how many appended blobs will be out per new appeding operation if the blob size is dynamic?

Thanks!

armink, 2021-09-09:
Hi, Grabber

The TSDB default storage mode is **rollover**. The oldest sector will clean if the database is full. You can see the TSDB control API: https://armink.github.io/FlashDB/#/api?id=control-tsdb

> How do I know how many appended blobs will be out per new appeding operation if the blob size is dynamic?

I don't understand this question.

Grabber, 2021-09-09:
> Hi, Grabber
> 
> The TSDB default storage mode is **rollover**. The oldest sector will clean if the database is full. You can see the TSDB control API: https://armink.github.io/FlashDB/#/api?id=control-tsdb

I was testing FDB_TSDB_CTRL_SET_ROLLOVER today, got it!

> 
> > How do I know how many appended blobs will be out per new appeding operation if the blob size is dynamic?
> 
> I don't understand this question.

Let's say the blobs have variable sizes and tsdb reaches the upper size limit. What happens? The oldest sector is cleaned or replaced entirely? Just the last blob from the oldest sector is replaced (doesn't make sense as in this example the blob size is variable)?

armink, 2021-09-09:
The blob max len is the `sec_size`. So it will return error when blob length large than `sec_size`.