armink/FlashDB #107: 请问如何获取tsdb发生回滚时的时间戳以及tsdb发生回滚时如何通知上层应用呢

illusionkiller, 2021-11-09:


armink, 2021-11-09:
没太看懂你的标题，可以举例详细说一下

illusionkiller, 2021-11-10:
就是当tsdb发生回滚时，好像应用层是无感知的，此时数据由于被回滚可能就查询不到了。
当发生tsdb回滚这种事件时，应用层是知道的并且可以做一些自己的处理，比如注册回滚事件回调处理类似于这样的机制。

armink, 2021-11-10:
明白了，那可以通过 control API 设置一个回滚前的回调函数，回滚的时候调用一下吧

illusionkiller, 2021-11-10:
好的，朱工，还有一个问题咨询一下，就是如何获取当前tsdb存储的最老的TSL和最新的TSL呢？
获取最新的数据我目前的做法是：先获取最新的时间戳（fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_GET_LAST_TIME, &cur_times);），然后根据获取到的时间戳进行查询最新TSL
获取存储的最老的TSL目前有什么方法吗？

armink, 2021-11-10:
把时间设置为 1970年1月1日 再去检索试试？

illusionkiller, 2021-11-10:
我的意思是能获取到的最老的数据，查看了源码后我发现当TSDB发生回滚后，随着TSL增加，最老的扇区也是变化的
`        /* calculate next sector address */
        if (sector->addr + db_sec_size(db) < db_max_size(db)) {
            new_sec_addr = sector->addr + db_sec_size(db);
        }
        else if (db->rollover) {
            new_sec_addr = 0;
        } else {
            /* not rollover */
            return FDB_SAVED_FULL;
        }
        read_sector_info(db, new_sec_addr, &db->cur_sec, false);
        if (sector->status != FDB_SECTOR_STORE_EMPTY) {
            /* calculate the oldest sector address */
            if (new_sec_addr + db_sec_size(db) < db_max_size(db)) {
                db->oldest_addr = new_sec_addr + db_sec_size(db);
            } else {
                db->oldest_addr = 0;
            }
            format_sector(db, new_sec_addr);
            read_sector_info(db, new_sec_addr, &db->cur_sec, false);
        }`
如果直接索引1970年1月1日肯定索引不到了。最老的可以索引到的TSL肯定是存到最老的扇区里面的，所以我能想到的做法就是先获取最老的扇区信息，然后根据扇区信息获取最老的TSL（即存在扇区里的第一条TSL）。找到了相关的接口（read_sector_info）和数据结构信息（tsdb_sec_info_t中的start_time成员），但是却无法直接使用

armink, 2021-11-11:
那感觉比较靠谱的方法，是跟 FDB_TSDB_CTRL_GET_LAST_TIME 类似，增加 FDB_TSDB_CTRL_GET_OLDEST_TIME 控制字

illusionkiller, 2021-11-11:
好的，我自己这边先修改一下实现测一测

armink, 2021-11-11:
好的，改完以后，欢迎提交 PR 哈