armink/FlashDB #108: fdb_tsl_append 追加数据写入失败，报错 FDB_SAVED_FULL

kenny-33, 2021-12-03:
遇到一个追加数据失败的问题，在 Zephyr 操作系统中，上电时使用如下设置进行初始化：

``` /* Flashdb 初始化. */
    sys_mutex_init(&m_fdb_mutex);
    fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_LOCK, (void *)lock);
    fdb_tsdb_control(&tsdb, FDB_TSDB_CTRL_SET_UNLOCK, (void *)unlock);

    fdb_err_t ret = fdb_tsdb_init(&tsdb, "fdb_ts", "fdb_tsdb1", get_time, 1054, NULL);
    __ASSERT(ret == FDB_NO_ERR, "fdb_tsdb_init() failed: %d", ret);
```

然后使用如下函数每隔 1S 追加一条数据，data 的长度在 1024Byte 左右.
```
    struct fdb_blob blob;
    fdb_err_t err = fdb_tsl_append(&tsdb, fdb_blob_make(&blob, data, len));
    LOG_DBG("flashdb write data: %d", err);
    if (err != FDB_NO_ERR) {
        if (err == FDB_SAVED_FULL) {
            LOG_WRN("flashdb write full");
        } else {
            LOG_ERR("flashdb write err: %d", err);
        }
    }
```

当我数据写满后，进入滚动状态，若此时设备掉电了，重新上电之后，会出现新的数据无法写入，一直报出 FDB_SAVED_FULL 提示。想请问是什么问题吗？

重新上电后每次追加写入时，都进入了函数 update_sec_status() 中的下面状态：
```
    else if (sector->status == FDB_SECTOR_STORE_FULL) {
        /* database full */
        return FDB_SAVED_FULL;
    }
```

使用的芯片型号为 MX25L256，存储空间是 32MByte(即 256Mbit)，下面是我的配置表：
```
#define FAL_PART_TABLE                                                             \
{                                                                                  \
    {FAL_PART_MAGIC_WORD, "fdb_kvdb1", NOR_FLASH_DEV_NAME, 0, 2*4096, 0},          \
    {FAL_PART_MAGIC_WROD, "fdb_tsdb1", NOR_FLASH_DEV_NAME, 0x400000, 200*4096, 0}, \
}
```

armink, 2021-12-04:
当前扇区存储满了以后，FDB 会自动找到下个扇区进行存储，这块的逻辑单步跟一下，看下下个扇区有找到吗

illusionkiller, 2021-12-08:
这个问题我也遇到了，现象与楼主差不多，数据发生回滚以后，此时复位，写入不成功，发现tsdb打印的日志中最老扇区和当前使用扇区与复位前不一致，最终查到是由于初始化函数中
[image]
此行代码导致，注释掉以后就没有问题了，以前的版本中是没有这行代码的，之前的版本中也测试过回滚后没有这种情况。不太明白朱工加入这行代码的目的是为了防止什么问题吗？现在我这边注释掉后才能保证回滚的正常使用。

armink, 2021-12-10:
大概知道问题了，当时是这次 提交 https://github.com/armink/FlashDB/commit/8114f881470fa3923adccaded8689b138778120a 引入的，目的是解决未启用回滚时，数据库满的问题，你按照下面的方式改进下，再试试呢

[image]

illusionkiller, 2021-12-11:
好的谢谢，有什么问题的话再进行反馈

armink, 2021-12-12:
@illusionkiller 方便提交一下 PR 不 ;>

kenny-33, 2021-12-15:
好的，谢谢解答！最近一直在忙其他的事情，没有及时关注这个

kenny-33, 2021-12-30:
改了之后昨晚测试了几次，发现还是会出现这个问题，特别是当存储满了之后再进行复位。就不会进行滚动了，一直报出 FDB_SAVED_FULL.

armink, 2021-12-30:
> 改了之后昨晚测试了几次，发现还是会出现这个问题，特别是当存储满了之后再进行复位。就不会进行滚动了，一直报出 FDB_SAVED_FULL.

是用的 master 最新版本吗？我有测试过你说的场景，看着是没问题了。

kenny-33, 2021-12-30:
其他部分没有更新，用的还是原来的代码，就按上述的改动了一下。。。

kenny-33, 2021-12-31:
更新之后目前没啥问题了，但是一个 FAL_PART_TABLE 的配置分区有最大限制吗？超过 8M 就初始化不成功。

armink, 2021-12-31:
@kenny-33 FAL 没有这个限制的

closed 2022-01-04 by kenny-33