armink/FlashDB #110: 关于使用kvdb过程中的一些疑问

illusionkiller, 2021-12-09:
朱工，有两个问题向您确认一下：
1、kvdb好像没有提供类似于tsdb清除的接口（目前自己添加了一个测试可用）
2、kvdb给统一个参数设置相同值和不同值时，在具体的操作步骤上查看源码好像是一样的，
[image]
此处加入一个去重机制，如果发现存储的值与设置的值相同，直接退出的话是不是会更好一些呢？
3、上一个Issues 我的解答中提出的疑问也麻烦您确认一下

armink, 2021-12-09:
> kvdb好像没有提供类似于tsdb清除的接口（目前自己添加了一个测试可用）

使用 fdb_kv_set_default 是不是就可以了

> kvdb给统一个参数设置相同值和不同值时，在具体的操作步骤上查看源码好像是一样的

这个之前 EasyFlash 是有的，已经那时缓存都在 RAM 里，对比也快。但现在 FDB 没有 RAM 缓存，所以每次对比都要执行 Flash 读取，会增加整体的修改时间的。

illusionkiller, 2021-12-10:
1、我的意思是清除所有参数不是恢复默认，set_default 好像达不到清除的效果
2、即便不改，find_kv 操作是少不了的，在find_kv后边加入比对，就可以避免后面几步操作，后面几步操作是比较费时间的吧

illusionkiller, 2021-12-10:
针对第二个问题我明白您的意思了，find_kv接口并无法直接返回KV值，还是需要一次读操作才能知道KV值，那就没有增加比对的必要了，目前我在应用层做了去重处理，尽可能减少IO操作，想着flashDB如果有的话，应用层就不需要做了，下个版本有考虑增加KV值的缓存的需求吗

armink, 2021-12-10:
暂时不会考虑加入 KV 值 的缓存，主要感觉产品上出现的概率比较低

illusionkiller, 2021-12-10:
好的，关于数据回滚后复位（或者重新上电）导致的写入失败问题，我在上个Issues中提出了一个解决方法（注释掉了初始化过程中新版本添加的一行代码）朱工麻烦您确认一下，这样修改有没有什么风险。

armink, 2021-12-10:
感谢反馈，回复你了哈

closed 2021-12-11 by illusionkiller