armink/FlashDB #113: 每次设置或获取kvbd值之前都需要初始化才可以吗

yeccc, 2021-12-30:
大神，请教个问题，每次调用函数 `fdb_kv_get_blob` 或 `fdb_kv_set_blob` 之前都需要调用函数 `fdb_kvdb_init`, 否则就会出现获取kvdb值失败，log: `get the 'boot_count' failed`, 不清楚具体是哪边有问题。

初始化：
在while(1) 主循环之前调用函数 `fdb_kvdb_init` 进行初始化，具体代码如下：
```
    struct fdb_default_kv default_kv;
    fdb_err_t result = FDB_NO_ERR;

    default_kv.kvs = default_kv_table;
    default_kv.num = sizeof(default_kv_table) / sizeof(default_kv_table[0]);                
    /* set the lock and unlock function if you want */
    fdb_kvdb_control(&test_kvdb, FDB_KVDB_CTRL_SET_LOCK, (void *)lock);
    fdb_kvdb_control(&test_kvdb, FDB_KVDB_CTRL_SET_UNLOCK, (void *)unlock);

    result = fdb_kvdb_init(&test_kvdb, "test_kv", "app", &default_kv, NULL);

    uart_printf("db init result:%x\r\n", result);
```

读写操作：
定时每隔3s 进行读写操作，此时只调用函数 `kvdb_basic_sample(&test_kvdb);`
这样子每次都会报错 `get the 'boot_count' failed`.

closed 2021-12-30 by yeccc

weicao0, 2023-09-13:
这个问题如何解决的?

weicao0, 2023-09-13:
这个问题如何解决的。我希望在开机时，一个进程完成数据库初始化，其他进程通过链接so库使用fdb_kv_get_blob函数读取数据。