armink/FlashDB #129: TSDB 循环覆盖问题

xun123456, 2022-01-12:
现象： 当TSDB 开始循环覆盖的时候 fdb_tsdb_init   初始化时 如果没有空扇区 。程序会把最后一个扇区的地址 赋给  tsdb  的  current using sector    当前使用扇区  ，会导致在便利读取数据的时候扇区地址出错

 
[image]

[image]

[image]

armink, 2022-01-12:
试试最新代码？这块有做更新 

https://github.com/armink/FlashDB/commit/83f1669773ea2497057cc78cf3f8c3bd1c70f991#diff-a4576c5e7b543ab64cbb612bd7b8f7ea25212334947e08e6897eca70f9511dc1

xun123456, 2022-01-13:
功能正常了