armink/FlashDB #130: LittleFS implementation

GiulioDallaVecchia, 2022-01-20:
Hi FlashDB developer team,

I'm having a look about this DB. It seams very good and during this days I'm trying to use it with my STM32 board. I'm wondering if I can add the possibility to use this DB with LittleFS file system. 

What do you think about?

Best regards,

Giulio Dalla Vecchia

armink, 2022-01-20:
Yes. The FlashDB can run on filesystem mode. You can see the linux demo.

GiulioDallaVecchia, 2022-01-21:
Hi @armink,

Thanks for the reply! I've written the functions to work with littleFS file system (open, read, write, erase). I suggest you to create a porting file for each file system, otherwise fdb_file.c become very long and complex.

Best regards,

Giulio Dalla Vecchia

armink, 2022-01-21:
The FDB filesystem mode support **POSIX** and **LIBC** API mode. Most filesystems support this type of API. So there is no need to modify the `fdb_file.c` file.

GiulioDallaVecchia, 2022-01-21:
In many embedded systems, it's used this file system [https://github.com/littlefs-project/littlefs]. It is very useful also because support static allocation. Unfortunately, it's not POSIX or LIBC API mode.

Best regards,
GDV

armink, 2022-01-21:
Maybe you can try the [RT-Thread RTOS](https://www.rt-thread.io/). It has a [Virtual File System](https://www.rt-thread.io/document/site/programming-manual/filesystem/filesystem/) and support POSIX or LIBC API mode.

The [littlefs](https://packages.rt-thread.org/en/search.html?search=littlefs) and [FlashDB](https://packages.rt-thread.org/en/search.html?search=FlashDB) as software package on the RT-Thread. There is no need to modify the code when using these two packages on any hardware platform.