armink/FlashDB #137: 关于fdb_file.c中使用open的错误

justjzp, 2022-03-18:
open属性中应该加入O_BINARY或者O_RAW二进制属性，否则在写入的数据中若包含0x0a,则会自动加上0x0d,最终导出读取的数据有一个字节异常

justjzp, 2022-03-18:
[image]
[image]
如图，测试每次写入的字节长度为32字节，如图红色标记，该处值应该为1，实际写入的也为1，由于数据包含0x0a的缘故，写入后自动补上0x0d，最终导致该数据被冲刷，读取数据的时候该值丢失，并且可能导致整个32字节数据错位

armink, 2022-03-18:
嗯，请问你是使用的什么文件系统？

方便提交一个 PR 增加 O_BINARY 参数吗

justjzp, 2022-03-18:
在window下使用mingw进行仿真测试，linux还未测试，晚点linux上测试一下，估计也是一样的效果，晚点提交一下linux测试结果，测试完成后再提交一个PR

justjzp, 2022-03-18:
linux下并没有O_BINARY和O_RAW属性，如果数据是二进制的话，建议还是使用FDB_USING_FILE_LIBC_MODE

armink, 2022-03-18:
好的，感谢测试

justjzp, 2022-03-18:
linux下默认就已经是二进制了，不会出现window下出现的0x0a前面补0x0d的问题，该问题可能只是windows系统下特有的问题

armink, 2022-03-18:
难怪呢，还是默认加上去吧