armink/FlashDB #142: Compile Error when not using FDB_KV_USING_CACHE

T721, 2022-04-12:
when FDB_KV_USING_CACHE is undefined to avoid caching of elements there is a compile error in fdb_kvdb.c:1486
in function check_and_recovery_kv_cb, where the following is used:

[image]

how can this be solved?

T721, 2022-04-12:
This is my temporary solution, as the check is only performed in startup.
Another option would be to also guard the method call with the defines but i'm not sure about side effects.
Anyone had the same problem?
[image]

armink, 2022-04-13:
Yes, FDB_KV_USING_CACHE cannot be disable at this time. Why do you want to disable it?

T721, 2022-04-13:
im using it on multiple cores/instances and do not want to deinit/init everytime. When i cache the elements and edit the same stuff on another instance, i need to deinit / init the first instance beforehand so there is not old data present.
For this reason i do not want to use chaching and set the following in fdb_cfg.h

[image]

which disables the following

[image]

armink, 2022-04-13:
Yes, would you like to fix this.

T721, 2022-04-14:
Hi @armink,
sure, i can have a look at this next week.

armink, 2022-04-15:
👍

armink, 2022-04-19:
You can fork first. Then start a PR from your repository.

closed 2022-04-25 by T721