armink/FlashDB #151: kvdb插入三万条记录的性能问题

roboidler, 2022-05-13:
hi
我们有个门禁卡的项目想使用flashdb,    我们想用kvdb来存储每个
门禁卡的信息，一台设备大约需要最多存储三万条数据。
我做了个简单的实验想在flash中插入三万条记录做测试:
    for (count = 1; count < 30000; count++) {
        snprintf(key,128,"%010d", count);
        snprintf(val,128,"2023-08-11 14:39 %8d", count);
        fdb_kv_set(&kvdb, key, val);
        LOG_D("insert [%d]", count);
    }
测试发现插入非常非常慢（我等了三四个小时三万条数据也没加完），而且会经常打印 GC XXXXX类似的信息
请问是我使用方式不对吗？ 另外还想请问，三万条数据中检索 key的性能如何？

Regards,

robo

armink, 2022-05-13:
这种最好分库来存

roboidler, 2022-05-13:
您指的分库来存是怎么实现？  是把三万条数据分多个 分区来存吗，  如果是这样，每个分区最多存多少个数据比较合适...
我现在的测试是。。一个分区，添加一两千条数据就会非常慢了..

armink, 2022-05-13:
是的，多个分区

Jason-Lipeixiang, 2022-05-13:
我也遇到这个问题了，fdb_kv_set写入新的键值特别慢，是哪里配置有问题吗？

slyant, 2022-06-08:
感觉超过100条，读取后插入的就越来越慢

shihang-zhang, 2022-06-15:
通过增大FDB_KV_CACHE_TABLE_SIZE定义的数值可以优化FlashDB的访问速度。 @armink 请问如果数据库里面定义的每个数据点都需要频繁访问，是否将FDB_KV_CACHE_TABLE_SIZE定义成数据点的个数可以达到最快的访问速度？谢谢！

hongxiong, 2022-06-16:
@armink FDB_KV_CACHE_TABLE_SIZE加大的确可以提高访问速度，不过也会加大RAM开销，且这个开销是每个分区都都会有一份cache table，是这样吧？
另外有个机制想确认一下：cache table是将最近**读或写**的目标更新到table中，这样看来，经常密集访问的目标会驻留在table中，并有效提高检索效率，这个理解是否正确？

armink, 2022-06-16:
> 通过增大FDB_KV_CACHE_TABLE_SIZE定义的数值可以优化FlashDB的访问速度。 @armink 请问如果数据库里面定义的每个数据点都需要频繁访问，是否将FDB_KV_CACHE_TABLE_SIZE定义成数据点的个数可以达到最快的访问速度？谢谢！

对的，可以增大 CACHE

armink, 2022-06-16:
> @armink FDB_KV_CACHE_TABLE_SIZE加大的确可以提高访问速度，不过也会加大RAM开销，且这个开销是每个分区都都会有一份cache table，是这样吧？ 另外有个机制想确认一下：cache table是将最近**读或写**的目标更新到table中，这样看来，经常密集访问的目标会驻留在table中，并有效提高检索效率，这个理解是否正确？

不是每个分区都有一个 Cache，数量是可以设定的。正如你的理解，FlashDB 里的 Cache 机制使用类 LRU 的算法，会将访问频率较高的目标，加大其保留在 Cache 中的权重。