armink/FlashDB #156: 希望可以优化在GC回收时其脏扇区的擦除策略!

SundayRX, 2022-06-14:
FlashDB在GC时花费了较长时间集中擦除多块脏扇区，在拥有较多扇区存储的配置下有可能造成卡顿!

10KB大小FLASH存储空间，扇区大小为1KB，使用KVDB只存储了一个键值，键值大小为300B左右，重复SetKeyValue之后各扇区逐步变为藏区，FLASHDB发现剩余存储扇区只剩下一块时触发了GC回收，GC集中把之前的7块脏区全部擦除，这无疑把最耗时的擦除工作全部堆积到了这次GC中，造成单次操作耗时过长。

实际应用中可能使用到512KB甚至1MB大小的区间作为平衡磨损空间，这样的情况发生时是高达1000次的循环擦除，所以能否改进GC回收机制，在某扇区变为脏区后立即将其擦除并标记为空扇区，这样可以均摊擦除时间而不是把擦除时间都集中在某个时间点擦除，从而有效减少单次GC导致的卡顿.

armink, 2022-06-14:
> FLASHDB发现剩余存储扇区只剩下一块时触发了GC回收

还剩多少空闲块，再进行 GC 的动作，这个是可以配置的，默认现在试试 1 ，要不你改大一些试试？

SundayRX, 2022-06-14:
感谢回复！！！

已将FDB_GC_EMPTY_SEC_THRESHOLD改大甚至为9 这样做的确降低了单次GC时擦除的扇区数但同样也提高了GC频次。
由于该值是对KVDB.C中的宏直接设置的，因此在多个数据库场景下，各数据库的存储扇区数一般并不相同，这个解决方案并不是很完美。

最后想问下朱工，设计FlashDB时使用剩余空扇区数阈值来触发GC而非使用”实时“GC，是出于对FLASH寿命和单次写操作花费时间的因素考虑吗？

还请多指教！！！

armink, 2022-06-14:
对的，所以攒到最后一起处理

SundayRX, 2022-06-14:
OK，已了解，那暂时先更改下我的存储刷新策略吧。

不过还是希望FlashDB后期能改进一下这个小小的问题。

armink, 2022-06-14:
这个也不能算是问题吧，或者你那边有没有其他的一些策略？

SundayRX, 2022-06-14:
可能我用词不太恰当，还请朱工不要太在意，仅仅是想让FlashDB变得更完美些。

将同步存储改为异步存储，可以使用队列存放要保存的数据然后单独开辟一个线程去做Flash存储任务，这样就不会影响用户线程。
或者改为仅在关机时存入Flash中（仅数据量较少时可用，且时间不易过长否则影响用户体验）。

armink, 2022-06-14:
嗯嗯，将来我是想着最好把 GC 接口开放出去，这样用户可以针对自己情况，择时进行手动 GC ，这样会不会好些呢？

SundayRX, 2022-06-15:
开放GC接口确实是一个不错的解决方案，对于我的应用场景来说，有一个另外的改动思路，可能工作量比较大,或许不太切合实际 朱工看下就好:

我注意到GC回收的类型有两种，一种是全脏扇区（全部废弃数据），另一种是半脏扇区（废弃数据+正在使用的数据），全脏扇区直接擦除标记即可，半脏扇区需要先进性数据移动，再进行擦除，有着更多的开销。
能否在SetKeyValue或者DelKeyValue这些操作函数中增加一个额外的GCOpt参数，该参数有三个枚举：

1 GC_Auto 由FlashDB自动进行垃圾回收  这个是FalshDB的默认GC策略
2 GC_Enforce1 强制GC模式1 只回收全脏的扇区 只进行擦除
3 GC_Enforce2 强制GC模式2 回收全脏和半脏的扇区 先做数据移动 再进行擦除 这个相当于强制触发一次FlashDB的默认GC

如果拥有这个功能，我只需要在每次存储操作的时候使用GC_Enforce1或者GC_Enforce2参数即可，其他用户使用默认GC_Auto参数则不受任何影响。

armink, 2022-06-15:
👍 这个建议挺好的，相当于给用户几种 GC 的策略可以供选择  

> GC_Enforce1 强制GC模式1 只回收全脏的扇区 只进行擦除

这种扇区出现的概率可能非常低呢，你这边大概是啥样的使用场景呢？

SundayRX, 2022-06-16:
之所以在拥有GC_Enforce2全回收模式下又有GC_Enforce1半回收模式 是考虑以下几点：

1 若只有GC_Enforce2模式则每次都进行全回收（全脏+半脏） 对于半脏区来说其触发回收的频率会过高 更多的消耗FLASH寿命
2 GC_Enforce2模式相比GC_Enforce1模式多了对半脏区的回收 这增加了单次操作的时间
3 GC_Enforce1模式主要应用于对于单值反复存储SetKeyValue的场景，这样扇区块均经历了从半脏区到全脏区的状态过度。若每次都只用GC_Enforce2模式则会导致擦除频率过高，扇区无法脱离半脏状态或因GC提前进入了全脏状态，使用GC_Enforce1模式则可让区块完全被使用到全脏区的状态再进行擦除，提高FLASH寿命

对于上面所提到的在SetKeyValue或者DelKeyValue这些操作函数中增加一个额外的GCOpt参数，后来经过思考考虑到：
1 增加新参数项目改动较大，工作量要考虑
2 老用户也需要对这些函数的传入参数进行修改，也很麻烦

所以新考虑增加fdb_GCCollect(fdb_db_t db, GCOpt Opt);函数，让有强制GC需求的用户在存储操作后自行调用这个GC回收函数就可以了，GCOpt参数仅有两个GC_Enforce1和GC_Enforce2。

以上方案仅为我个人想法，可能有些许不足！

armink, 2022-06-16:
对于 `单值反复存储SetKeyValue的场景` 是不是可以将这些 KV 独立一个 数据库，不要影响项目在用的其他改动频率低的 KV ，这样这个高频写入的数据库就使用 GC_Auto 就可以了呢？

SundayRX, 2022-06-17:
其实这个单值本身就被放在了一个独立的KV数据库中，只不过依赖于均衡磨损来增加可写入次数，因此这个独立数据库的大小很大导致单个数据库的管理扇区很多，扇区多就意味着如果仅仅使用GC_Auto 依然会产生集中擦除较多扇区引起的卡顿问题。

GC_Auto 平时不进行任何操作 仍依赖于FlashDB在写入时发现可写入扇区低于阈值时进行GC回收

armink, 2022-06-18:
先确认下你的诉求，我感觉你需要的像是要给 FlashDB 增加 3 个功能

- 1、Block 增加一个全脏的模式（以前只有半脏），支持优先格式化全脏 Block 作为空闲的 Block
- 2、格式化全脏 Block 的时机不要等到都满了再格式化，会占用较大时间，如果发现最好就立刻格式化，降低格式化粒度
- 3、格式化完的 Block 的使用时机按照格式化顺序进行使用，不能刚格式化完就立刻使用，避免磨损均衡问题

我这么理解的对吗？

SundayRX, 2022-06-20:
朱工理解应该没错了 还是麻烦您了！

armink, 2022-06-20:
这些功能你可以搞定吗 :>

closed 2022-06-20 by SundayRX