armink/FlashDB #161: kv_hdr_data占用flash空间比例太大

guoweilkd, 2022-06-30:
```c
struct kv_hdr_data {
    uint8_t status_table[KV_STATUS_TABLE_SIZE];  /**< KV node status, @see fdb_kv_status_t */
    uint32_t magic;                              /**< magic word(`K`, `V`, `4`, `0`) */
    uint32_t len;                                /**< KV node total length (header + name + value), must align by FDB_WRITE_GRAN */
    uint32_t crc32;                              /**< KV node crc32(name_len + data_len + name + value) */
    uint8_t name_len;                            /**< name length */
    uint32_t value_len;                          /**< value length */
};
```
STM32中，kv_hdr_data成员间4字节对齐。假如FDB_WRITE_GRAN=8，则 kv_hdr_data  占7*4=32byte.kv_hdr 是直接存到flash中的，是否占用flash空间比例有点太大了?

lilealyh, 2022-08-17:
我也遇到同样的问题，这个结构体改成1字节对齐（#pragma pack(1)），占用空间可以减小到22字节。

littlewhitezkz, 2024-10-18:
改小点