armink/FlashDB #162: TSDB queries out data that has been marked for deletion

Joiey, 2022-07-01:


armink, 2022-07-02:
@chwncy 合入 #158 后这个确实是个问题呢

我先把 PR 回退了，要不在 read_kv 中增加参数？

chwncy, 2022-07-02:
@armink 提交的代码确实出了问题，会导致#160的问题，原因是我在提交代码的时候，直接把上面红框里面的代码拷贝到了下面
[image]

这就会把crc_is_ok赋值为false;
在read_sector_info函数中会判断crc_is_ok为false, status在不等于FDB_KV_PRE_WRITE 和 FDB_KV_ERR_HDR 时，就会打印CRC32 check failed.

因此我把crc_is_ok赋值为true即可解决，不知armink 觉得可行否？

ps：在我实际运用中，基于258kflash测试，以每次写入相同键值的4字节数据，将其写满，重新上电运行，如果没有屏蔽状态标记为删除的情况下，总共需要花费12秒左右，若屏蔽，则将仅花费6秒左右。

armink, 2022-07-02:
把 38 行改为 true 可以的

armink, 2022-07-02:
有两种情况会用到 read_kv 

- 一个是开机加载 数据库 时，此时确实不需要读取已删除 KV 
- 另外时迭代 KV 时，此时用户也可能迭代已删除的 KV 

针对这些场景，可以把 read_kv 功能再好好抽象一下

chwncy, 2022-07-02:
> 把 38 行改为 true 可以的，但为何要返回读取错误呢

因为在当前文件中，read_kv函数为static函数，而当前文件在调用read_kv的时候并没有使用到read_kv的返回值，因此return将不存在意义，考虑到后期开发，此处可以改成return FDB_NO_ERR比较好一些。

chwncy, 2022-07-02:
> 有两种情况会用到 read_kv
> 
> * 一个是开机加载 数据库 时，此时确实不需要读取已删除 KV
> * 另外时迭代 KV 时，此时用户也可能迭代已删除的 KV
> 
> 针对这些场景，可以把 read_kv 功能再好好抽象一下

针对第二种情况，我在使用中并未触及到，对于如果不需要迭代已删除的数据时，可以考虑跳过校验标记为删除的值，此方案就给其他用户上电节省时间做为一种参考，不在提交PR了。

junzimengyou, 2022-07-05:
#160 的确把crc_is_ok=true能解决这个问题，已经重新测试了。

Joiey, 2022-08-02:
谢谢，但是时序数据库的情况呢