armink/FlashDB #165: 请教下在每分钟都需要存三十几条记录的场景下推荐怎么使用

Quiet-Dev, 2022-07-13:
看下了以前的讨论，当记录数较多时使用kv方式特别慢，这样的话如果使用ts方式是否能快速的增删查找，业务所限，每天要求存4w多条记录。

Quiet-Dev, 2022-07-19:
> 4w多条记录强烈建议换方案吧，换日志型数据库

请问有哪个可以推荐下吗，不能用只读的

junzimengyou, 2022-07-19:
您是什么硬件方案？

落花繁华
***@***.***

Quiet-Dev, 2022-07-19:
> 您是什么硬件方案？ 落花繁华 ***@***.***  
> […](#)

N32G455的MCU, 512K ROM , 144K RAM,  用的旺宏的8MB spi nor flash

junzimengyou, 2022-07-19:
LittleFS 
https://github.com/littlefs-project/littlefs

落花繁华
***@***.***

  
您是什么硬件方案？ 落花繁华 @.***  
 …

Quiet-Dev, 2022-07-19:
> LittleFS  https://github.com/littlefs-project/littlefs 落花繁华 ***@***.***  
> […](#)

lfs也进行了尝试，但是在文件数量较多时，发现删除的速度比较慢，我尝试一次性删除一个具备100字节数据的60个文件（目录层级较多），发现需要5s左右才能结束，这影响了运行效率，而实际上删除在特定的情况下每小时要删除一千多记录。听说使用少量大文件的方式可以增加效率，但是这并不符合预期。

armink, 2022-07-19:
分多个区分多个数据库再试试

junzimengyou, 2022-07-19:
spi flash可能不适合这种运用场景，建议使用emmc或者tf卡，自带磨损均衡，但是fatfs文件系统不支持掉电保护，如果频繁擦写的话，我以前用的文件系统ext4来搞的，spi flash还是不适合大数据量频繁擦写的。

落花繁华
***@***.***

  
LittleFS  https://github.com/littlefs-project/littlefs 落花繁华 @.***  
 …

ZanoZ, 2023-02-03:
> 

lfs对小文件和大文件的处理方式是不一样的，所以你如果没其他办法，可以试试合并小文件为大文件，然后不适用删除，而是标记无效的方式处理,另外使用其他进程进行无效数据的整理的删除动作。
对你这个应用，好好规划文件存储的形式，可能更为重要，而不是将功能压力放到文件系统上，毕竟嵌入式文件系统的设计都相对薄弱。