armink/FlashDB #167: fdb_write_gran应该怎么理解？

as564335sa, 2022-07-22:
大佬，fdb_write_gran 这个宏应该怎么理解？它的作用到底是什么啊，和实际flash的页，扇区、单次写入大小有什么关联？看源码也没找到有什么关联呢，它是怎么确定的

armink, 2022-08-05:
这个是 Flash 的 `写粒度` ，即写 Flash 时的最小单位

benpicco, 2023-02-25:
What does a setting of 1 mean? Even on RAM one can't address individual bits - is `write_gran` in meant to mean Bytes instead?

armink, 2023-02-26:
> What does a setting of 1 mean? Even on RAM one can't address individual bits - is `write_gran` in meant to mean Bytes instead?

Yes, Flash and RAM are addressed by byte. After the NorFlash is erased, each bit can be individually modified to 0. In FlashDB the write granularity is 1 bit. For example: address 0, write 0xFE first, then write 0xFC.

benpicco, 2023-02-26:
Ok, but it looks like `.write_gran` is not used anywhere outside of one demo application.
Is this intentional, is it only a hit to the application and not used by FlashDB at all?

armink, 2023-02-27:
The `write_gran` must be defined in fdb_cfg.h https://github.com/armink/FlashDB/blob/master/inc/fdb_cfg.h#L32

kleinsand, 2023-11-22:
> 这个是 Flash 的 写粒度 ，即写 Flash 时的最小单位
如果芯片没有uint8，uint8都被typedef成了uint16，每次写入都是写入16bit，这个值fdb_write_gran是设置成8吗