armink/FlashDB #169: Document upgrade and downgrade processes

zamazan4ik, 2022-08-04:
Hi.

After reading the documentation it is not clear for me, how upgrade and downgrade processes should be performed over the database. Could you please describe somewhere in the documentation:

- How to upgrade the database to the newer version (hopefully with no downtime)
- How to downgrade the database to the older version (hopefully with no downtime)
- Maybe some compatibility policies and notes between versions

Thanks in advance!

armink, 2022-08-05:
Hi. Which capability is the database upgrade? Can you give an example?

zamazan4ik, 2022-08-05:
Sure. With a "upgrade" word I mean upgrading FlashDB version to the newer version, e.g. 1.0 -> 1.1. "Downgrade" is the same but in a vice versa direction: 1.1 -> 1.0.

Why this information is important for the users? Because if the upgrade/downgrade processes are complicated (or even not supported without full reexport), it could be much harder for the users perform their usual regular upgrades (and downgrades, if something goes wrong with the newer database version).

armink, 2022-08-05:
OK, I known. The database format of FlashDB from V1.0 to V1.1 are fully compatible. It's just a functional enhancement. If there is an incompatible version, I will write it in the documentation.