armink/FlashDB #171: Trigger a GC check after alloc KV failed.

lilealyh, 2022-08-11:
[16:41:36.942] [FlashDB][kv][historyDB] (..\..\..\src\app\r_pi\database\FlashDB-v1.1.1\src\fdb_kvdb.c:997) Trigger a GC check after alloc KV failed.
[FlashDB][kv][historyDB] (..\..\..\src\app\r_pi\database\FlashDB-v1.1.1\src\fdb_kvdb.c:1166) Warning: Alloc an KV (size 54) failed when new KV. Now will GC then retry.
[FlashDB][kv][historyDB] (..\..\..\src\app\r_pi\database\FlashDB-v1.1.1\src\fdb_kvdb.c:1250) The remain empty sector is 1, GC threshold is 1.
[FlashDB][kv][historyDB] (..\..\..\src\app\r_pi\database\FlashDB-v1.1.1\src\fdb_kvdb.c:997) Trigger a GC check after alloc KV failed.
[FlashDB][kv][historyDB] (..\..\..\src\app\r_pi\database\FlashDB-v1.1.1\src\fdb_kvdb.c:1174) Error: Alloc an KV (size 54) failed after GC. KV full.
key:1660204997

//500ms调用一次下面的代码没过多久就出现了上面的打印
trainingData_t wdata; //结构体大小为20字节
itoa(globleTimeStamp,string,10);
wdata.startStamp = globleTimeStamp;
wdata.jumpCount = j;
wdata.jumpTime = j++;
makeHistoryTest(string,(char*)&wdata,sizeof(trainingData_t));

void makeHistoryTest(const char *key,void *value,uint16_t valueLength)
{
    struct fdb_blob blob;
    fdb_kv_set_blob(&kvdb, key, fdb_blob_make(&blob, value, valueLength));
    rt_kprintf("key:%s\n", key);
}
我的需求是每秒存入sizeof(trainingData_t) 大小的数据，key取四字节的UNIX时间戳，value就是这个结构体数据。请大神指导一下，谢谢

edwardhey, 2022-11-28:
我目前也遇到这个问题，同求指导，谢谢

edwardhey, 2022-11-28:
> [16:41:36.942] [FlashDB][kv][historyDB] (......\src\app\r_pi\database\FlashDB-v1.1.1\src\fdb_kvdb.c:997) Trigger a GC check after alloc KV failed. [FlashDB][kv][historyDB] [...]

我也遇到这个问题了，请问你解决了么？