armink/FlashDB #180: 关于时间戳类型为int32，数据库可迭代最大时间为2038-01-19 11:14:07

ousugo, 2022-09-30:
现在 `fdb_tsl_iter_by_time` 这个 API 中传入的开始时间戳和结束时间戳定义为 `fdb_time_t`，即 `int32_t`。这个类型的最大值为  2,147,483,647。
[image]

 2147483647 对应的时间为：2038-01-19 11:14:07。所以超过这个时间的 TSL 就无法被迭代器读取到。
[image]

想请教下我这样理解是对的吧，我代码实际测试也是如此，查询 2037 的数据没有问题，查询 2039 的数据就读不到。

armink, 2022-09-30:
可以开启 `FDB_USING_TIMESTAMP_64BIT` 宏，使用 64 位时间戳

ousugo, 2022-10-12:
启用 FDB_USING_TIMESTAMP_64BIT 宏以后，tsl 的时间戳和实际存储的时间戳一直对不上。tsl 的时间戳在 `1089977456` 附近，RTC 设置的时间是 2022 年。然后查询迭代的起始时间戳包括了 `1089977456` 也查询不到对应的 TSL。
[image]

还没仔细研究是为什么，先记录一下。