armink/FlashDB #184: 一个分区，能使用多个数据库吗？

GarMingLi, 2022-10-20:
```
**
 * The time series database initialization.
 *
 * @param db database object
 * @param name database name
 * @param path FAL mode: partition name, file mode: database saved directory path
 * @param get_time get current time function
 * @param max_len maximum length of each log
 * @param user_data user data
 *
 * @return result
 */
fdb_err_t fdb_tsdb_init(fdb_tsdb_t db, const char *name, const char *path, fdb_get_time get_time, size_t max_len, void *user_data)

/* partition table */
#define FAL_PART_TABLE                                                                                                 \
    {                                                                                                                  \
        {FAL_PART_MAGIC_WORD, "history", NOR_FLASH_DEV_NAME, 0, 4 * 1024 * 1024, 0},                               \
    }

fdb_tsdb_init(&test_tsdb1, "db1", "history", get_time, 128, NULL) ；
fdb_tsdb_init(&test_tsdb2, "db2", "history", get_time, 128, NULL) 
fdb_tsdb_init(&test_tsdb3, "db3", "history", get_time, 128, NULL) 
fdb_tsdb_init(&test_tsdb4, "db4", "history", get_time, 128, NULL) 

```
比如我一个分区有4M，能不能初始化把该分区分成多个数据库？分区大小固定，迭代的时候，分别迭代四个库效率高还是迭代一个库效率高？

GarMingLi, 2022-10-22:
> 

时序数据库，首次初始化，需要格式化，耗时比较久。我使用外部nor flash，给该时序数据库分配4M的空间，格式化时候要几十秒，这样正常吗？

GarMingLi, 2022-10-23:
Nor Flash 擦除时候，32K、64K擦除速度会比4K擦除快些。所以格式化的时候，可否先把整个分区先擦除，然后flash驱动根据分区大小选择64K、32K、4K的方式逐步擦除flash，然后再格式化？不是每4K擦除再格式化

GarMingLi, 2022-10-23:
或者有没有必要在初始化的时候格式化，能不能在用到的时候再格式化某扇区？

GarMingLi, 2022-10-23:
> 或者有没有必要在初始化的时候格式化，能不能在用到的时候再格式化某扇区？ “FDB_TSDB_CTRL_SET_NOT_FORMAT”设置这个命令，影响使用吗？

实测不能用。不格式化，“init_ok”为false，“fdb_tsl_append”的时候就返回“FDB_INIT_FAILED”退出了

ZanoZ, 2023-02-09:
我的理解，格式化分区，就能根据数据库格式(主要是记录长度)实现/建立快速定位数据记录起始地址的功能
在这个角度，一个分区只能搞一个数据库。
坏处就是，不同数据库写入更新频率不一样的时候，就是不同分区磨损会有区别。。
这个就需要根据具体写入情况斟酌分区大小，尽量实现均衡磨损。

不知道我这样说对不对