armink/FlashDB #187: [建议]能够增加一个异步模式，单独开启一个线程进行gc与Collect操作？

wdfk-prog, 2022-11-19:
# 原因
- 使用flashdb的kvdb数据库，使用文件模式。
- 发现写入数据库时，若触发了gc或Collect操作，会在当前写入数据库的线程进行此操作。
- 如果线程分配堆栈不足，会导致堆栈溢出，系统卡死。
## 例如
- 1秒写入一次当前时间戳至kvdb数据库中，分配这个线程的堆栈为2048
- 没出现问题时，占用率最大81%.但是触发了Collect操作时，堆栈就不够了，系统卡死
- 无法打印日志提示问题所在，导致难以排查。
- 后续增加线程堆栈为4096解决，占用率为51%
# 建议
- 能否增加一个异步模式来专门执行gc与Collect操作？
- 就像ulog的异步线程的方式那样。
- 写入数据库并不要求实时性，只需要确保数据的准确就好了。
- 异步方式需要一个缓冲区，缓冲区满了才能传入数据，可能导致缓冲区未满时掉电，导致数据缺失了一部分。
- 但是这个情况，可以添加一个函数，立刻刷新缓冲区。然后给用户使用。 用户自己实现掉电保存与硬件。
- 所以可以增加一个异步线程的选项，提供用户使用。看用户是选择实时，还是效率

wdfk-prog, 2022-11-19:
另外： 文件模式设置的扇区数量如原来为2，现在设置为5，是否需要重新格式化？我操作好像会卡死
从5扇区设置为2扇区也一样

wdfk-prog, 2022-11-19:
还有就是 cmd命令show只能查看当前数据的存储地址与数据大小。没法查看数据内容。
另一个是文件保存的内容如何解析。我将文件传输至pc查看16进制文件，无法知晓具体内容，只能知道个大概。

ZanoZ, 2023-02-09:
楼主提出了一个看似复杂但是实际又合理的需求，以下是个人观点：
1，首先要注意这是数据库，不是文件
2，异步会增加同步操作的复杂性，导致对操作系统依赖性更强，不同的操作系统规则不同，这不利于代码维护。
3，没了。
楼主提出了一个看似复杂但是实际又合理的需求，以下是个人观点：
1，首先要注意这是数据库，不是文件
2，异步会增加同步操作的复杂性，导致对操作系统依赖性更强，不同的操作系统规则不同，这不利于代码维护。
3，可以在每次启动init的时候进行一次或者2次主动gc，这样会不会好一点

wdfk-prog, 2023-02-13:
- 数据库也有规律,起码我自己能够阅读处理存储的数据内容.不然在排查问题的时候,无法得知存储的数据.
- 但是现在的同步执行,对线程栈的要求还是蛮大的.如果栈容量小了,GC操作容易爆栈,而且GC执行的时间并不固定,难以排查.
- init的时候进行一次或者2次主动gc,看起来像是饮鸩止渴.