armink/FlashDB #195: 调大FDB_GC_EMPTY_SEC_THRESHOLD后，部分Flash sector一直没有被使用

shihang-zhang, 2023-02-01:
Hi @armink, 因为GC耗时过长，我们想通过调整FDB_GC_EMPTY_SEC_THRESHOLD的设置，让GC的时候擦除的块变少，从而减小GC的时间。但是发现下面的问题。
实验过程中不停的更改数据，总共128个sector，threadhold 设置为125。发现FlashDB只对sector0，sector1和sector2 有擦除操作，其它块没有操作。等于剩余的块没有参与操作。
请问我们可以通过什么办法，减小GC的耗时，最后只擦除相对少的块。
多谢！

armink, 2023-02-01:
频繁的更改数据，应该不会增加较多的 Flash 占用，所以只有 sector0，sector1和sector2 有擦除情况应该是正常的，可能你的数据量本身就比较小

你可以试试，频繁的 新增+修改 数据

shihang-zhang, 2023-02-06:
@armink,是的 我们的数据点个数是定值，数量比较少，不会动态增加或减少。所有的数据点加在一起不会超过一个sector的大小。但是每个数据点都会被频繁修改，这样的需求使我们想通过增加sector的个数来增加平衡磨损使用的空间，使每个数据点可以达到百万次的修改。现在如果使用默认FDB_GC_EMPTY_SEC_THRESHOLD值，可以让所有的sector都被使用到，但是GC的时间过长，达不到系统的要求。增大FDB_GC_EMPTY_SEC_THRESHOLD值，GC时间减少了，但是会有部分sector永远不会被用到，减小平衡磨损的空间，从而达不到数据点修改次数的需求。

armink, 2023-02-07:
这个问题目前看起来在当前版本可能是无解的，GC 的过程在 KVDB 中必不可少。

另外，你也可以确认下 GC 时，是哪个阶段的耗时过长，再针对性的想一些优化手段。

shihang-zhang, 2023-02-07:
@armink 就是GC时会集中擦除Flash的各个块，擦除各个块的时间累加起来就比较长，如果只擦除俩块三块sector，耗时就相对不明显。通过调大FDB_GC_EMPTY_SEC_THRESHOLD，可以达到控制GC擦除块的个数。我读了一下代码，如果在每次iteration过程中不从第一个块开始，应该是可以让后面的块有机会被使用到。是否可以在header里面添加块擦除的信息？或者开辟一些系统kv参数来保存这些信息？这样在程序运行的流程中通过这些信息做平衡磨损操作。

hongxiong, 2023-02-19:
@armink 你好，我和 @shihang-zhang 是同一个项目的同事，非常喜欢你的FlashDB的功能和轻量且易用。
目前我们在实际使用时，的确存在这样的用例：数据量不大但是改写很频繁，所以我们选择预留较多Flash sectors并且调大FDB_GC_EMPTY_SEC_THRESHOLD，目的是：尽量均匀磨损各sector并且使得GC每次清理较少数目的sector以保证系统不会在GC时阻塞太久。
比如我们现在总体persistent数据<10KB左右，但改写很频繁，考虑Flash生命周期和磨损次数，我们预留了20个sector（每个sector 2KB）给FlashDB分区，然后设定FDB_GC_EMPTY_SEC_THRESHOLD=10，目前实验结果显示10个以后的sector可能永远不会被用到。
请您帮忙证实一下并提供解决思路，我们可以贡献这部分优化。

armink, 2023-02-24:
> @armink 就是GC时会集中擦除Flash的各个块，擦除各个块的时间累加起来就比较长，如果只擦除俩块三块sector，耗时就相对不明显。通过调大FDB_GC_EMPTY_SEC_THRESHOLD，可以达到控制GC擦除块的个数。我读了一下代码，如果在每次iteration过程中不从第一个块开始，应该是可以让后面的块有机会被使用到。是否可以在header里面添加块擦除的信息？或者开辟一些系统kv参数来保存这些信息？这样在程序运行的流程中通过这些信息做平衡磨损操作。

你的这个场景感觉好优化，相当于高频次的修改某几个 KV ，此时确实瓶颈是在 擦除时间 。

增加一种局部 GC 的模式功能即可，每次 GC 出来的空闲空间，够当前的 KV 长度即可停止 GC

armink, 2023-02-24:
> @armink 你好，我和 @shihang-zhang 是同一个项目的同事，非常喜欢你的FlashDB的功能和轻量且易用。 目前我们在实际使用时，的确存在这样的用例：数据量不大但是改写很频繁，所以我们选择预留较多Flash [...]

你好，感谢支持，感觉局部 GC 可能就可以了，你们请先评估一下

Johnxjj, 2024-04-16:
空闲块分配做成随机的就好了，别做成顺序分配。

Johnxjj, 2024-04-16:
FDB_GC_EMPTY_SEC_THRESHOLD我想问这个能设置成0吗，我现在只想分配2个扇区给flashDB