armink/FlashDB #199: Single file

GiulioDallaVecchia, 2023-02-13:
Hi all,

is it possible to use FlashDB with a single file?

armink, 2023-02-13:
Yes, you can use single file to simlate the Flash storage. Like this driver file https://github.com/armink/FlashDBAutoTestBSP/blob/master/applications/flash_sim.c . It was used in FlashDB auto test CI.

GiulioDallaVecchia, 2023-02-14:
Thanks @armink,

if I use an external flash, I'll set FDB_WRITE_GRAN = 1, but I have to check if the flash is erased before writing operation?

armink, 2023-02-15:
The FlashDB will auto erase the flash before write operation.

The external flash demo you can reference this project https://github.com/armink/FlashDB/tree/master/demos/stm32f405rg_spi_flash

GiulioDallaVecchia, 2023-02-15:
Hi @armink,

thanks a lot for sharing. I have a problem, every time I reboot the board and fdb_kv_init is called, all Kv status is equal to FDB_KV_DELETED and all keys is set to the default value. What is the problem?

GiulioDallaVecchia, 2023-02-16:
Sorry @armink, 

It was a my fault due to an error on external flash driver.

Thanks a lot!

closed 2023-02-16 by GiulioDallaVecchia