armink/FlashDB #212: 初始化正常，数据写入正常，读取时校验异常，读取时CRC校验失败

syh501, 2023-05-08:
移植FlashDB-1.1.2到高拓6441平台，给了一段分区大小为32k的分区给到FlashDB管理，初始化正常，数据写入看起来也正常成功，但是读取的时候失败，跟踪了一下，好像是CRC校验失败了，没有什么头绪，还望有空了帮忙看看~
（另外，感觉好像初始化的时候，格式化了2次）
部分运行日志：
[I/FAL] Flash Abstraction Layer (V0.5.99) initialize success.
partition name:[user_data] [user_data]
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1912) KVDB size is 32768 bytes.
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00000000).
erase addr:0x00000000 len:4096
erase addr:0x00603000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00001000).
erase addr:0x00001000 len:4096
erase addr:0x00604000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00002000).
erase addr:0x00002000 len:4096
erase addr:0x00605000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00003000).
erase addr:0x00003000 len:4096
erase addr:0x00606000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00004000).
erase addr:0x00004000 len:4096
erase addr:0x00607000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00005000).
erase addr:0x00005000 len:4096
erase addr:0x00608000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00006000).
erase addr:0x00006000 len:4096
erase addr:0x00609000
[FlashDB][kv][data] (/home/uwen/matter/atbm_4m_matter/src//extern_src/FlashDB-1.1.2/src/fdb_kvdb.c:1698) Sector header info is incorrect. Auto format this sector (0x00007000).
erase addr:0x00007000 len:4096
erase addr:0x0060A000
[FlashDB][kv][data] All sector header is incorrect. Set it to default.
erase addr:0x00000000 len:4096
erase addr:0x00603000
erase addr:0x00001000 len:4096
erase addr:0x00604000
erase addr:0x00002000 len:4096
erase addr:0x00605000
erase addr:0x00003000 len:4096
erase addr:0x00606000
erase addr:0x00004000 len:4096
erase addr:0x00607000
erase addr:0x00005000 len:4096
erase addr:0x00608000
erase addr:0x00006000 len:4096
erase addr:0x00609000
erase addr:0x00007000 len:4096
erase addr:0x0060A000
[FlashDB] FlashDB V1.1.2 is initialize success.
[FlashDB] You can get the latest version on https://github.com/armink/FlashDB .
fdb_kvdb_init result:0
mode: next generation
size: 112/28672 bytes.
[FlashDB][sample][kvdb][basic] ==================== kvdb_basic_sample ====================
[FlashDB][sample][kvdb][basic] get the 'boot_count' failed

armink, 2023-05-08:
FDB_WRITE_GRAN 设置了多大？

syh501, 2023-05-08:
> FDB_WRITE_GRAN 设置了多大？

设置的为1
#define FDB_WRITE_GRAN 1

const struct fal_flash_dev atbm_onchip_flash =
{
    .name = "atbm_onchip_flash",
    .addr = FLASH_USR_DATA_SECTION2_ADDR,
    .len = FLASH_USR_DATA_SECTION2_LEN,
    .blk_size = DEVICE_SECTOR_SIZE,
    .ops = {init, read, write, erase},
    .write_gran = 1
};

armink, 2023-05-08:
我感觉还是 Flash 驱动的问题比较大，这块要重点检查一下

syh501, 2023-05-08:
> 我感觉还是 Flash 驱动的问题比较大，这块要重点检查一下

好的好的，非常感谢，因为这块被封装起来了，只给了一些上层的接口，有可能是这个原因，我再尝试解决一下，非常非常感谢！

closed 2023-05-08 by syh501