armink/FlashDB #227: fdb_tsl_iter_by_time 函数异常

DarryZh, 2023-08-09:
https://github.com/armink/FlashDB/blob/master/src/fdb_tsdb.c文中的的search_start_tsl_addr函数tsl.addr.index = start + FDB_ALIGN((end - start) / 2, LOG_IDX_DATA_SIZE);

armink, 2023-08-14:
没有太看懂你的意思，方便再描述详细些吗？也包括你设置的 ALIGN 是多大

DarryZh, 2023-08-15:
我的配置文件中FDB_WRITE_GRAN是8，即1byte；

https://github.com/armink/FlashDB/blob/master/src/fdb_tsdb.c 中search_start_tsl_addr函数，计算 tsl.addr.index = start + FDB_ALIGN((end - start) / 2, LOG_IDX_DATA_SIZE) 时异常。
输入参数为start:48，end:68, LOG_IDX_DATA_SIZE为20, 所以有FDB_ALIGN((end - start) / 2, LOG_IDX_DATA_SIZE)计算出来为12, 进一步tsl.addr.index = start + 12, 结果为60。

但我的tsdb的log所在的正确位置为tsl.addr.index = start，即48。如果描述不清楚或者错误，还请不吝赐教。

DarryZh, 2023-08-15:
进一步的原因，我认为是FDB_ALIGN 宏定义的实现出了问题，第二个参数 align如果不为2^n，则会出现奇怪的对齐数据，这是我下面做的最小测试的结果：

FDB_ALIGN(10, 10) : 18
FDB_ALIGN(10, 20) : 12
FDB_ALIGN(20, 20) : 36
FDB_ALIGN(20, 10) : 20
FDB_ALIGN(10, 8) : 16
FDB_ALIGN(32, 16) : 32

armink, 2023-08-15:
确实是的呢，你方便优化下这块代码吗？

DarryZh, 2023-08-17:
https://github.com/armink/FlashDB/pull/233

DarryZh, 2023-08-17:
已解决

closed 2023-08-17 by DarryZh