armink/FlashDB #230: 功能建议：TSDB添加主键Key索引

muguou, 2023-08-14:
目前flashDB的tsdb数据库存入数据使用append函数，很简便。
但是取出数据的方式完全依靠  遍历+回调函数实现。
在c语言里回调函数实在是说不上好用，不知道能否提供一种方式来便利的直接获取对应点存储的数据。
本人水平有限给不出什么建设性建议，以下是我的一点简单设计。
能否在TSDB中添加主键Key这个选项，底层数据格式是 uint32_t。
在查询数据时，会返回一个数组，数组中存放符合要求的数据 主键Key
TSDB的API添加一个根据Key获取tsl或者blob数据的函数，这样对于嵌入式软件程序员来说，更符合使用直觉。

armink, 2023-08-14:
TSDB 的主键 KEY 是时间，目前接口因为是回调模式，所以可能不太好理解

你也可以自行改造下，传入一个 buf 取代回调，出参也是这个 buf

muguou, 2023-08-15:
> TSDB 的主键 KEY 是时间，目前接口因为是回调模式，所以可能不太好理解
> 
> 你也可以自行改造下，传入一个 buf 取代回调，出参也是这个 buf

作者大大你好，我主要是想着能像队列一样使用tsdb。

我对tsdb的主要使用方式就是和操作队列一样。
最后位置 append + 最开始位置read 就是我对tsdb的所有操作了。
目前的方式，我每次都只能通过 遍历+回调+修改tsl 状态来取最开始没被读过的数据，相当于每次都遍历了许多没用的数据，这样的方式我感觉稍微有点低效了。本质上是因为遍历加回调的结果没办法 存储下来为下一次检索提供帮助，随机访问肯定是实现不了的，但是如果想要通过这个tsl去链接下一个tsl不知道是否可以实现，没太仔细研究过，我看api函数里没有对单tsl级别的操作函数。
入参出参的改造倒是可以实现，但是依靠目前的api还是需要去遍历的，如果要实现我上边说的效果是不是只能深入flashdb的实现代码去改造了

armink, 2023-08-15:
用 fdb_tsl_to_blob  及   fdb_blob_read     是可以单独读取 TSL ，但还不允许在外部写入